stm32duino/Arduino_Core_STM32 #1709: Arduino CLI fails to build with standard parameters

zfields, 2022-05-04:
The Arduino CLI compiles most firmware using the following parameter and an `.ino` file.

```none
-b, --fqbn string                  Fully Qualified Board Name, e.g.: arduino:avr:uno
```

For example:

```none
arduino-cli compile -b arduino:avr:uno /home/user/Arduino/MySketch.ino
```

If I try this with an ST board I get the following error:

```
$ arduino-cli compile --fqbn STMicroelectronics:stm32:GenL4 /home/user/Arduino/MySketch.ino
In file included from /root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.2.0/cores/arduino/stm32/stm32_def.h:48,
                 from /root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.2.0/cores/arduino/stm32/clock.h:19,
                 from /root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.2.0/cores/arduino/wiring_time.h:23,
                 from /root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.2.0/cores/arduino/wiring.h:38,
                 from /root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.2.0/cores/arduino/Arduino.h:36,
                 from /tmp/arduino-sketch-43EA3F8101597CC7E1F859A3B3B56A0F/sketch/MySketch.ino.cpp:1:
/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.2.0/system/Drivers/CMSIS/Device/ST/STM32L4xx/Include/stm32l4xx.h:175:3: error: #error "Please select first the target STM32L4xx device used in your application (in stm32l4xx.h file)"
  175 |  #error "Please select first the target STM32L4xx device used in your application (in stm32l4xx.h file)"
      |   ^~~~~

Error during build: exit status 1
```

I compiled the firmware in the Arduino IDE and I noticed the `--fqbn` used is more complex.

```
-fqbn=STMicroelectronics:stm32:BluesW:pnum=SWAN_R5,upload_method=swdMethod,xserial=generic,usb=none,xusb=FS,opt=osstd,dbg=none,rtlib=nano
```

I tried to compile using the Arduino CLI with the more complex `--fqbn` parameter, but I'm still hitting the same issue.

```
$ arduino-cli compile --fqbn STMicroelectronics:stm32:BluesW:pnum=SWAN_R5,upload_method=swdMethod,xserial=generic,usb=none,xusb=FS,opt=osstd,dbg=none,rtlib=nano /home/user/Arduino/MySketch.ino
In file included from /root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.2.0/cores/arduino/stm32/stm32_def.h:48,
                 from /root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.2.0/cores/arduino/stm32/clock.h:19,
                 from /root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.2.0/cores/arduino/wiring_time.h:23,
                 from /root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.2.0/cores/arduino/wiring.h:38,
                 from /root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.2.0/cores/arduino/Arduino.h:36,
                 from /tmp/arduino-sketch-43EA3F8101597CC7E1F859A3B3B56A0F/sketch/MySketch.ino.cpp:1:
/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.2.0/system/Drivers/CMSIS/Device/ST/STM32L4xx/Include/stm32l4xx.h:175:3: error: #error "Please select first the target STM32L4xx device used in your application (in stm32l4xx.h file)"
  175 |  #error "Please select first the target STM32L4xx device used in your application (in stm32l4xx.h file)"
      |   ^~~~~

Error during build: exit status 1
```

What am I missing? I have compiled sketches (including this one) for nearly every platform I can think of, but I cannot get ST to compile.

fpistm, 2022-05-04:
Hi @zfields 
I can tell you it works as all CI uses `arduino-cli` to build all targets of the core and I use it daily to build and tests.

For STM32 core, `pnum` is requested else it could not build. Other options could be ignored if you build with default one.
Else, simply add it. example for enable USB:
`STMicroelectronics:stm32:BluesW:pnum=SWAN_R5,usb=CDCgen `

Which arduino-cli version you used?

zfields, 2022-05-04:
That is great news! So that means I must be missing something in my environment I suppose?

I'm using the latest version of the [Arduino CLI v0.21.1](https://github.com/arduino/arduino-cli/releases/tag/0.21.1), running it on Linux.

I've prepared a Dockerfile, so you can see my exact environment:

```Dockerfile
# Copyright 2022 Blues Inc.  All rights reserved.
# Use of this source code is governed by licenses granted by the
# copyright holder including that found in the LICENSE file.

# Build development environment
# docker build --file Dockerfile.arduino-cli --tag arduino-buildpack .

# Launch development environment
# docker run --entrypoint bash --interactive --rm --tty --volume "$(pwd)":/host-volume/ --workdir /host-volume/ arduino-buildpack

# Define global arguments
ARG DEBIAN_FRONTEND="noninteractive"
ARG UID=1000
ARG USER="blues"

# POSIX compatible (Linux/Unix) base image
FROM debian:stable-slim

# Import global arguments
ARG DEBIAN_FRONTEND

# Define local arguments
ARG ARDUINO_CLI_VERSION=0.21.1
ARG BINDIR=/usr/local/bin
ARG ECHO_BC_FILE='$bcfile'

# Define environment variables
ENV ARDUINO_UPDATER_ENABLE_NOTIFICATION=false

# Establish development environment
RUN ["dash", "-c", "\
    apt-get update --quiet \
 && apt-get install --assume-yes --no-install-recommends --quiet \
     bash \
     ca-certificates \
     curl \
     python-is-python3 \
     python3 \
     python3-pip \
     ssh \
 && pip install \
     pyserial \
 && apt-get clean \
 && apt-get purge \
 && rm -rf /var/lib/apt/lists/* /tmp/* /var/tmp/* \
"]

# Download/Install Arduino CLI
RUN ["dash", "-c", "\
    curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh | BINDIR=${BINDIR} sh -s ${ARDUINO_CLI_VERSION} \
# && arduino-cli completion bash > /usr/share/bash-completion/completions/arduino-cli.sh \
 && mkdir -p /etc/bash_completion.d/ \
 && arduino-cli completion bash > /etc/bash_completion.d/arduino-cli.sh \
 && echo >> /etc/bash.bashrc \
# && echo \"for bcfile in /usr/share/bash-completion/completions/* ; do\" >> /etc/bash.bashrc \
# && echo \"for bcfile in /etc/bash_completion.d/* ; do\" >> /etc/bash.bashrc \
# && echo \"    [ -f \\\"${ECHO_BC_FILE}\\\" ] && . \\\"${ECHO_BC_FILE}\\\"\" >> /etc/bash.bashrc \
# && echo \"done\" >> /etc/bash.bashrc \
"]

# Configure Arduino CLI
RUN ["dash", "-c", "\
    arduino-cli config init \
 && arduino-cli config add board_manager.additional_urls \
     https://raw.githubusercontent.com/stm32duino/BoardManagerFiles/main/package_stmicroelectronics_index.json \
     https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json \
     https://raw.githubusercontent.com/sparkfun/Arduino_Apollo3/main/package_sparkfun_apollo3_index.json \
     https://raw.githubusercontent.com/adafruit/arduino-board-index/gh-pages/package_adafruit_index.json \
     https://github.com/earlephilhower/arduino-pico/releases/download/global/package_rp2040_index.json \
 && arduino-cli core update-index \
 && arduino-cli core install arduino:avr \
 && arduino-cli core install arduino:mbed_nano \
 && arduino-cli core install arduino:samd \
 && arduino-cli core install STMicroelectronics:stm32 \
 && arduino-cli core install esp32:esp32 \
 && arduino-cli core install SparkFun:apollo3 \
 && arduino-cli core install adafruit:nrf52 \
 && arduino-cli core install adafruit:samd \
 && arduino-cli core install rp2040:rp2040 \
 && arduino-cli lib install \"Blues Wireless Notecard\" \
"]

ENTRYPOINT ["arduino-cli"]

CMD ["help"]
```

fpistm, 2022-05-05:
> That is great news! So that means I must be missing something in my environment I suppose?

Probably.

> If I understand correctly, you are saying that I can compile (in the most basic sense) using `STMicroelectronics:stm32:BluesW:pnum=SWAN_R5`?

Yes. It has been tested on Linux and Windows with ardunio-cli 0.21.1.

> I've prepared a Dockerfile, so you can see my exact environment:

OK but I've no idea about this. GHA also uses docker and it works fine. 
I advise you to install it locally and retry. 

GHA:
https://github.com/stm32duino/Arduino_Core_STM32/runs/6291574962?check_suite_focus=true#step:4:323

On my Linux host:
```
arduino-cli compile --fqbn=STMicroelectronics:stm32:BluesW:pnum=SWAN_R5,usb=CDCgen /local/data/ide/arduino/arduino-latest/examples/01.Basics/AnalogReadSerial/AnalogReadSerial.ino 

Sketch uses 30016 bytes (1%) of program storage space. Maximum is 2097152 bytes.

Global variables use 3936 bytes (0%) of dynamic memory, leaving 651424 bytes for local variables. Maximum is 655360 bytes.
```

zfields, 2022-05-05:
Okay, I over-simplified my example to make it easier to share. I can confirm, when using the most basic command syntax you provided, the `arduino-cli` will compile the sketch with the Dockerfile supplied above. :+1: 

This works:
```
arduino-cli compile -b STMicroelectronics:stm32:BluesW:pnum=SWAN_R5 /home/user/Arduino/MySketch.ino
```

In my actual use case, I'm trying to ensure our examples compile without warning on our ST hardware, and I want to fail the build if a warning is encountered. So I added two additional build arguments:

- `--warnings all`
- `--build-property build.extra_flags=-Werror`

Adding them back one at a time, I was able to narrow it down to `--build-property build.extra_flags=-Werror`.

By adding this one parameter to the build it causes everyone of my ST boards to fail in a similar fashion, given a message like the one below.

```
In file included from /root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.2.0/cores/arduino/stm32/stm32_def.h:48,
                 from /root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.2.0/cores/arduino/stm32/clock.h:19,
                 from /root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.2.0/cores/arduino/wiring_time.h:23,
                 from /root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.2.0/cores/arduino/wiring.h:38,
                 from /root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.2.0/cores/arduino/Arduino.h:36,
                 from /tmp/arduino-sketch-43EA3F8101597CC7E1F859A3B3B56A0F/sketch/MySketch.ino.cpp:1:
/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.2.0/system/Drivers/CMSIS/Device/ST/STM32L4xx/Include/stm32l4xx.h:175:3: error: #error "Please select first the target STM32L4xx device used in your application (in stm32l4xx.h file)"
  175 |  #error "Please select first the target STM32L4xx device used in your application (in stm32l4xx.h file)"
      |   ^~~~~

Error during build: exit status 1
```

This is bizarre, because the build does not produce any warnings when `--warnings all` is specified.

So I guess the name of this bug should be, "Arduino CLI fails to build when parameter `--build-property build.extra_flags=-Werror` is specified".

This only occurs when attempting to build using the ST platform. I have been able to compile for the esp32, nrf52, apollo, mbed and avr platforms without issue.

fpistm, 2022-05-06:
Hi @zfields 
First, why you don't tell us you used that before?
If you do not provide us all information we (you and me) are losing time to find why you got issue.

You redefine a property which is defined in the boards.txt:
https://github.com/stm32duino/Arduino_Core_STM32/blob/949ecc80d0399b4a5ead733ff57be5f728a47c4d/boards.txt#L5814

So that is normal it fails. An issue exists on arduino-cli to avoid such issue:
https://github.com/arduino/arduino-cli/issues/846

Extract:
> 
> However, the developer often has need for the recipe-specific properties, and so uses them in boards.txt ([example](https://github.com/MCUdude/MightyCore/blob/v2.0.5/avr/boards.txt#L103-L111)). This presents the problem that the user risks inadvertently overriding the flags defined by the boards platform if they use the "extra flags" properties to modify the compilation command.
> 
> A solution for this conflict could be to establish two sets of the properties: one set of properties for the developer, and another for the user. These should be clearly documented as being dedicated to the sole use by either the platform developer or the user.

In your case you can define it properly,(I didn't test):

`--build-property build.extra_flags=-Werror -D{build.product_line} {build.enable_usb} {build.xSerial}`

or you can use the `build_opt.h` to `-Werror` to the command line.

zfields, 2022-05-06:
Oh wow, that's unreal. Sorry for not sharing the full detail in the first place, I couldn't imagine that would be the problem.

For the sake of CI, I would like to have a single formation of the compile command and substitute { fqbn } and { sketch }.

I would like to learn more about the other approach. Where is `build_opt.h`? I am unfamiliar with this file. Can you point me toward any documentation to support this approach?

fpistm, 2022-05-06:
Here:
https://github.com/stm32duino/wiki/wiki/Customize-build-options-using-build_opt.h

Anyway I do not advise to use -Werror as probably HAL will raised some warnings which will be converted as error but as we use HAL from the Cube "as is", no issue about this will be accepted here. For those warnings, they have to be submitted to the official repo on https://github.com/STMicroelectronics/.

I close this issue as it is invalid and it is not the place to discuss arduino-cli options.

zfields, 2022-05-21:
Sharing the results (for others who may find this) of my continued research and conversations with the Arduino CLI maintainer, @per1234.

Based on the flags I was attempting to send, `-Werror` and `-Wno-unused-parameter`, I should have been using the build property `compiler.cpp.extra_flags`.

The property I originally tried to modify, `build.extra_flags`, is reserved for the platform vendor (e.g. ST), while `compiler.cpp.extra_flags` has been set aside for end-user modification.

Modifying `compiler.cpp.extra_flags` is the officially supported path for the Arduino CLI, and serves the same purpose as `build_opt.h` (which is ST specific).

fpistm, 2022-05-23:
Hi @zfields,

thinking about that I made the PR #1725 to redefine some of the build properties.

PR #1725 (2022-05-25): feat: allows arduino-cli to override build property
Arduino-cli allows to override some standard Arduino build property with a custom value.
Previously some of them was already used in `boards.txt`/`platform.txt` preventing to use it.

This PR remove this usage by defining some custom properties making the default one empty and usable by end user.
`build.extra_flags`, `compiler.c.extra_flags` and `compiler.S.extra_flags` can now be redefined all other were already empty by default.

Fixes #1709
closed 2022-05-25 by fpistm