stm32duino/Arduino_Core_STM32 #1738: Not achieving pin interrupt 26khz

darshansm, 2022-06-09:
I am generating an external interrupt around 26khz. But ISR is running around 22khz. I tried with different GPIOS.

But same thing is working in 
https://github.com/GrumpyOldPizza/arduino-STM32L4
Board files. I don't know what is missing here 

**Board **
NUCLEO L476RG

ABOSTM, 2022-06-09:
Hi @darshansm,
Can you tell us more about your setup:
Which Arduino_Core_STM32 version ?
Which variant do you use ?
What is you sketch to generate external interrupt ? (please provide a minimal sketch reproducing the issue)
What is your measurement method ?
...

darshansm, 2022-06-09:
Arduino_Core_STM32 - 2.2.0
Varient -    STM32L4xx/L475R(C-E-G)T_L476R(C-E-G)T_L486RGT

Code -
const byte outPin = 13;
const byte interruptPin = PC3;

void setup() {
  pinMode(outPin, OUTPUT);
  pinMode(interruptPin, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(interruptPin), ISRfunc, RAISING);
}

void loop() {
 
}

void ISRfunc() {
  digitalWrite(outPin, HIGH);
delayMicroseconds(5);
digitalWrite(outPin, LOW);
}

Interrupt pin generates interrupt in 26k

ABOSTM, 2022-06-09:
In order to speed up the ISR processing, can you try the following:
```
const PinName outPinName = digitalPinToPinName(13);

void ISRfunc() {
digitalWriteFast(outPinName, HIGH);
delayMicroseconds(5);
digitalWriteFast(outPinName, LOW);
}
```

darshansm, 2022-06-09:
I tried that. It didn't help

ABOSTM, 2022-06-13:
I tested my Nucleo L476RG, with Core v2.2.0,
I used your sketch (I replace RAISING by RISING to fix compilation issue). How does this compile on your side ??
I used variant Nucleo L476RG, 
I used an external signal generator @26KHz  (PC3)
I uses an logical Analyzer to get the output frequency (D13)

It works perfectly, the output signal is 26KHz
[image]

ABOSTM, 2022-06-13:
So difference between your setup and mine is around signal generator and logical analyzer.
So please check that part.
Can you detail your setup ?

closed 2022-06-20 by fpistm