stm32duino/Arduino_Core_STM32 #1740: 【((us - ((us > 0) ? 1 : 0)) * tickPerMs)】might overflow when us is greater than 25565

MapleHan, 2022-06-14:
https://github.com/stm32duino/Arduino_Core_STM32/blob/b71582003800ebc5e550cb692fef80016cc43f58/cores/arduino/wiring_time.h#L76

fpistm, 2022-06-15:
Hi @MapleHan 
value depends of the system clock config.
Anyway, this function is not designed to be used with this value.
As stated by the official Arduino API reference:
https://www.arduino.cc/reference/en/language/functions/time/delaymicroseconds/

> Currently, the largest value that will produce an accurate delay is 16383; larger values can produce an extremely short delay. This could change in future Arduino releases. For delays longer than a few thousand microseconds, you should use delay() instead.

fpistm, 2022-06-15:
Moreover adding overflow management will decrease `delayMicrosecond` precision for small value.

closed 2022-06-20 by fpistm