stm32duino/Arduino_Core_STM32 #1741: IbusSensor does not work on stm32 version 2.0.0+

stryjekryjek, 2022-06-16:
Hello and welcome,
I have such a problem, namely I discovered that the ibusbm library does not work on the new version of stm32duino.
When I install version 1.9.0, everything works fine, but when I try to update to the new version 2.0.0, 2.1.0, 2.2.0, unfortunately it does not work.
Could I ask someone to solve the problem in my spare time ;) Thank you in advance for your help and for your fantastic work so far ;)

```

#include <IBusBM.h>

IBusBM IBusSensor;

HardwareSerial Serial2(PA3, PA2); //RX, TX

  
void setup() {

 
  IBusSensor.begin(Serial2);
  IBusSensor.addSensor(IBUS_MEAS_TYPE_GPS_DIST);

IBusSensor.setSensorMeasurement(1, 200); 

}

void loop() 
{

}
```

fpistm, 2022-06-16:
Could you be more precise on your setup? Boards? options? Ide version?...

Based on below link, I guess it is for F4 serie, like NUCLEO-F401RE

For ref:
https://www.stm32duino.com/viewtopic.php?f=7&t=1584

stryjekryjek, 2022-06-16:
I tested on:
Stm32F401ccu6 (Blackpill)
Stm32F411ceu6 (Blackpill)
NUCLEO-F401RE Stm32F401RE

Ide all version 1.8.9-1.8.19

Same situation on all configurations

Going back to stm32duino version 1.9.0 fixes the problem on all configurations.

fpistm, 2022-06-16:
Well quickly look to the library, it seems really not properly ported to support this core:
https://github.com/bmellink/IBusBM/blob/master/src/IBusBM.h#L44-L47

Several issue here:

1.  `_VARIANT_ARDUINO_STM32_` is only defined by Roger Clark core:

https://github.com/rogerclarkmelbourne/Arduino_STM32/search?q=_VARIANT_ARDUINO_STM32_

This core provides the standard definition required by Arduino: `ARDUINO_CORE_{arch}` so here it is `ARDUINO_CORE_STM32`
2. While the first check is for other core, the second one uses `STM32_CORE_VERSION`   is defined by this core. Moreover you stated it does not work for version > 1.9.0 while the error message states it is only for version higher.
3. Finally  the latest release here: https://github.com/bmellink/IBusBM/releases/tag/v1.1.5

> The original STM32 code was created for an old version of the STM32 core lib (before v1.6). This generated compiler errors for newer versions. The library now supports the current v1.9 core library version. No changes for any of the other supported architectures (avr, esp32, mbed)

This is clearly not coherent with point 2 which state it is only compatible with version >= 0x01090000.

So, with all those incoherent points, I guess issue is on library side and should be fixed there at least uses proper definition and check if it build properly.
Moreover we do not had hardware to test iBus feature so could not help more on this issue.