stm32duino/Arduino_Core_STM32 #1742: BUG, G491 MCU TIM7_DAC_IRQn not defined

Eddddddddy, 2022-06-19:
I added a new stm32 model, G491, but encountered a problem when compiling.
TIM7_DAC_IRQn not defined

https://github.com/stm32duino/Arduino_Core_STM32/blob/b71582003800ebc5e550cb692fef80016cc43f58/cores/arduino/stm32/timer.h#L98-L100

Only those MCU have TIM7_DAC_IRQn. It not have TIM7_DAC_IRQn like G491. 
[image]

I search the HAL repo.

https://github.com/STMicroelectronics/STM32CubeG4/blob/5b804dfaaeeafe5e87705b62099f420f822f6b28/Drivers/CMSIS/Device/ST/STM32G4xx/Include/stm32g491xx.h#L131

It only has TIM7_IRQn.

ABOSTM, 2022-06-20:
Hi @Eddddddddy ,
Thanks for reporting this issue,
I proposed a fix : #1744
Feel free to test it.

closed 2022-06-21 by fpistm

Eddddddddy, 2022-07-18:
Thank you so much for this fix. I try the v2.3.0 release. It can be compiled. 

But there is an issue about USB and Serial. Basically, I try to use Serial through USB. I choose "Enable ( generic 'Serial' )" and "CDC ( generic 'Serial' supersede U(S)ART )", and add "Serial.begin(115200);" in my code, but there is nothing showing in the serial port.

I will put this in a new issue.

thanks again

Eddddddddy, 2022-07-18:
okay, I see. The answer is, Arduino doesn't upload the firmware to the chip. I use usb dfu to upload. My system is an M1 Pro MacBook running macOS 12.4. I also tried STM32CubeProg, and I can't upload any firmware normally using dfu. But swd works fine. But this frustrates me a lot because my current design doesn't support swd.

fpistm, 2022-07-19:
DFU should work but you have to enter manuall in the bootloader mode.
See https://www.st.com/resource/en/application_note/cd00167594-stm32-microcontroller-system-memory-boot-mode-stmicroelectronics.pdf
For G491xx:
Pattern 15:
[image]

BL configuration:
[image]

commit 05dbfe95 (2022-06-21): Fix: CMSIS stm32g491 and stm32g4a1 add missing aliases for TIM7 and COMP4/5/6

Internal ticket 130525
Fixes #1742