stm32duino/Arduino_Core_STM32 #1758: .hpp file ending mandatory?

MasteringTheMess, 2022-07-12:
Hi all,

I ran into much trouble trying to compile a codebase that I already had to compile on STM32duino (using sloeber - but that is not part of the problem) some weeks (or was it already months?) ago.
Project got laid aside, I wanted to resume just to find out that I get weird error messages from the compiler.
After some fiddling around I could nail it down on a header file time.h that holds a small, simple class with a method to calculate system running time and convert it to a human-readable format.
Compiler complains that the type "class" is not known.
I the updated to latest 2.3.0 (was on 2.2.0) and the behaviour persists.
The only thing that works is renaming the header file to time.hpp
I have tested this only in a small example project that has nothing else than empty setup() and loop() as well as the #include time.h(pp)
Tested on sloeber and standard Arduino IDE.

[rant]
Since I am on windows, testing is very, very time consuming since every rebuilt after changing only smallest bits leads to a complete rebuild (thankfully not in sloeber!) which takes hilarious amount of time (5-10 minutes)
[/rant]

As to now I am quite sure that all it takes really is to rename the file to .hpp

This is quite strange, since I am almost 100% sure I changed nothing to my setup in the time between "codebase compiles" and "codebase annoys compiler and loads of errors are thrown"

MasteringTheMess, 2022-07-13:
Hello @fpistm,

sorry for not being precise, time.h is part of my own codebase.

fpistm, 2022-07-13:
OK so what do you expect?

closed 2022-07-13 by fpistm

MasteringTheMess, 2022-07-13:
What I expect?
Some help in understanding what is going on... and hopefully a way to resolve my problem

well, in the past my Time.h got included correctly, now the compiler complains about the "class" keyword.
I don't think that this is a problem with ambiguous includes

fpistm, 2022-07-13:
You should go to the forum for this kind of issue and I answered you. Using same name than standard include is not good practice which leads to strange issue. Why it works before I can't tell moreover without any code. About subject: .hpp ending mandatory. Answer is no. 😉

MasteringTheMess, 2022-07-13:
Thank you for the advice to go to the forum!

I thought this was an issue of the stm32duino, that's why I first came here. I remember changing my time.h from struct to class because the compiler complained about something the AVR compiler gracefully ignored...
In a way the STM32 compiler helps me write better code :)

MasteringTheMess, 2022-07-13:
I just keep thinking about this...

Is there a way of finding out what filenames are "taken"?
Or is it visible in the compiler output somewhere?

fpistm, 2022-07-13:
My guess is that time.h from the standard C library is simply not included but your one is used instead but when standard time.h is required it is probably to build c file with gcc. But as your time.h used class keyword it's failed. I don't understand why you simply not change the name of your file like this include will be done properly... This would be the same if you create stdio.h, math.h, string.h,... 
https://www.tutorialspoint.com/c_standard_library/time_h.htm

MasteringTheMess, 2022-07-13:
Of course I did this, my question is to be able to avoid such a problem in the future...
Maybe there exists a convention, that every library file (be it standard C, ARM core, STM32 core) is starts with lower case, so if I use upper case, I will always be safe?

MasteringTheMess, 2022-07-13:
maybe move this whole topic to a discussion?