stm32duino/Arduino_Core_STM32 #1774: wrong error management in i2C functions?

camelator, 2022-07-24:
I have an issue I don't know where it comes from... After some seconds, the I2C line remains high... but calling I2C functions always return good results.

During debugging my problem I can see that the following function  returns I2C_OK although the I2C line is crashed:
 if the result of the first receive call is not HAL_OK, it returns I2C_OK because there is no else block attached to the first 'if' , instead of returning an error code.

i2c_status_e i2c_master_read(i2c_t *obj, uint8_t dev_address, uint8_t *data, uint16_t size)
{
  i2c_status_e ret = I2C_OK;
  uint32_t tickstart = HAL_GetTick();
  uint32_t delta = 0;
  uint32_t err = 0;

#if defined(I2C_OTHER_FRAME)
  uint32_t XferOptions = obj->handle.XferOptions; // save XferOptions value, because handle can be modified by HAL, which cause issue in case of NACK from slave
#endif

#if defined(I2C_OTHER_FRAME)
  if (HAL_I2C_Master_Seq_Receive_IT(&(obj->handle), dev_address, data, size, XferOptions) == HAL_OK) {
#else
  if (HAL_I2C_Master_Receive_IT(&(obj->handle), dev_address, data, size) == HAL_OK) {
#endif
    // wait for transfer completion
    while ((HAL_I2C_GetState(&(obj->handle)) != HAL_I2C_STATE_READY) && (delta < I2C_TIMEOUT_TICK)) {
      delta = (HAL_GetTick() - tickstart);
      if (HAL_I2C_GetError(&(obj->handle)) != HAL_I2C_ERROR_NONE) {
        break;
      }
    }

    err = HAL_I2C_GetError(&(obj->handle));
    if ((delta >= I2C_TIMEOUT_TICK)
        || ((err & HAL_I2C_ERROR_TIMEOUT) == HAL_I2C_ERROR_TIMEOUT)) {
      ret = I2C_TIMEOUT;
    } else {
      if ((err & HAL_I2C_ERROR_AF) == HAL_I2C_ERROR_AF) {
        ret = I2C_NACK_DATA;
      } else if (err != HAL_I2C_ERROR_NONE) {
        ret = I2C_ERROR;
      }
    }
  }
  return ret;
}

fpistm, 2022-07-26:
Hi @camelator 

I've made a fix. Could you try it with your use case?

Thanks

camelator, 2022-07-26:
I can't see your changes. Where did you put it? Which branch?

fpistm, 2022-07-26:
And it is referenced by the issue 😉

closed 2022-07-29 by fpistm

camelator, 2022-08-14:
Sorry for the delay, but to me the fix is nott complete:
as I wrote, the function "request_from" should be changed as well, as it can read data although wrong data sent. So the data may be inconsistent.

PR #1775 (2022-07-29): fix(wire): ensure i2c bus is ready
Previously, i2c_master_write and i2c_master_read returned I2C_OK if first HAL call returned HAL_BUSY which was not correct.
Now make sure the i2c is ready, which guarantees a good initialization of the read or write sequence.

**Validation**

* [x] i2c scanner 
* [x] i2c memory device (eeprom)
* [x] temperature sensor (LM75B)
* [x] Master Reader Writer /  Slave Sender Receiver

Fixes #1774
commit e3808a29 (2022-07-29): fix(wire): ensure i2c bus is ready

Previously, i2c_master_write and i2c_master_read returned I2C_OK if
first HAL call returned HAL_BUSY which was not correct.
Now make sure the i2c is ready, which guarantees a good
initialization of the read or write sequence.

Fixes #1774