stm32duino/Arduino_Core_STM32 #1782: Continuous flashing in dfu mode is not working

darshansm, 2022-08-06:
**Core Version:** 2.3.0
**Board:** NUCLEO L476RG

But using upload protocol *dfu*. Everytime I need to make Boot pin high to enter dfu mode. Generally to enter dfu mode we open the com port with 1200 baud. It is not working. I tried in windows and Linux platform both.

Any workaround for this issue??

ABOSTM, 2022-08-08:
Hi @darshansm,
I don't understand what you mean by `Generally to enter dfu mode` ? other core version ? other STM32 MCU/board ? Other non STM32 MCUboards ?

From Application note AN2606 Rev 55,
§64  `STM32L47xxx/48xxx devices bootloader`
STM32L47xxx supports bootloader "pattern 7"
[image]
So, having to set Boot0 pin is a valid choice,
and it sounds normal to me, to set it each time you want to enter BootLoader.

And then, when USB enumerates  (when USB cable is plugged), it starts in DFU. see AN3156.

In order to be able to open COM port requires to be in DFU (and not reverse)

darshansm, 2022-08-08:
I Agree on that. But If I enable USB CDC, I can see the USB COM port. Using that COM port, how can we enter to dfu mode. Generally we can't change the Boot0 pin always. 

Previously I was using https://github.com/GrumpyOldPizza/arduino-STM32L4

Board package. There if I enable USB CDC and if I open and close COM port with 1200 baud rate it get entering in to dfu mode.

ABOSTM, 2022-08-08:
I am not familiar with GrumpyOldPizza.
What I told you is valid for STM32 DFU (ROM code), and it is mandatory because at reset, MCU needs Boot0 pin status to decide to continue in normal mode or in BootLoader mode.
Maybe GrumpyOldPizza uses an 2nd stage bootloader, 
meaning MCU goes only once in bootloader mode and install a FW which is a 2nd level bootloader.
Then even if you run in normal mode (from MCU point of view and no need for BOOT pin change), the code executed allows to flash the real application in Flash. Note: in such case 2nd level bootloader reserve some space in Flash (which is not usable for application).
This seems to be the case as I see this:
https://github.com/GrumpyOldPizza/arduino-STM32L4/blob/ac659033eadd50cfe001ba1590a1362b2d87bb76/variants/STM32L433CC-Butterfly/linker_scripts/STM32L433CC_FLASH.ld#L72

This core Arduino_Core_STM32, doesn't provided such 2nd level bootloader, but it is ready to use some existing ones like HID Bootloader, Maple , ...
[image]

fpistm, 2022-08-23:
Hi, 
automatic reset in dfu mode is not supported yet.
A PR was started several years ago: #710
Anyway it will require some more work.

closed 2022-08-23 by fpistm