stm32duino/Arduino_Core_STM32 #1786: random() without parameter like in standard Arduino API is missing.

ArminJo, 2022-08-06:
Standard Arduino programs do not compile if they use` random();`

E.g.
```
/home/runner/Arduino/libraries/NeoPatterns/src/NeoPatterns.hpp:1474:31: error: no matching function for call to 'random()'
         tRandom.Long = random();
                               ^
/home/runner/Arduino/libraries/NeoPatterns/src/NeoPatterns.hpp:1474:31: note: candidates are:
...
/home/runner/.arduino15/packages/stm32duino/hardware/STM32F1/2021.5.31/cores/maple/wirish_math.h:50:6: note: long int random(long int)
 long random(long max);
```

if I replace the line 50 in wirish_math.h
`long random(long max);`
with
`long random(long max = __LONG_MAX__);`
everything works fine.

Do you require a PR?

fpistm, 2022-08-07:
Hi,
Based on the log: `STM32F1/2021.5.31/cores/maple`
you used this core: https://github.com/rogerclarkmelbourne/Arduino_STM32/
not this one.

closed 2022-08-07 by fpistm

ArminJo, 2022-08-07:
Sorry you are right, it was the stm23duino in your path that confused me.