stm32duino/Arduino_Core_STM32 #1789: Mixing core_debug and Serial printing results in slowdown and lost output

matthijskooijman, 2022-08-10:
**Describe the bug**
When using both `core_debug()` and `Serial.println()` for writing to the serial port, some conflict occurs. `Serial` output becomes slow, and `core_debug()` output gets lost.

**To Reproduce**
```
$ cat STM32SerialAndVcoreDebug.ino 
#include <core_debug.h>

void setup() {
        Serial.begin(115200);
        Serial.println("0123456789");
        core_debug("FOOBARBAZ");
}

void loop() {
}
```

Compiled with:

    $ arduino-cli compile -v --warnings all --upload --fqbn STMicroelectronics-git:stm32:Nucleo_64:pnum=NUCLEO_WL55JC1,dbg=enable_log,opt=ogstd,upload_method=swdMethod

This results in:

[image]

**Expected behavior**
Both messages should be printed directly.

**Environment:**
 - OS: Linux
 - Arduino IDE version: arduino-cli git 2dd8976683f4a9f83fcee478620f1e83d11111b2 (0.25.1 with a few more commits)
 - STM32 core version: git e3808a29a49aeb57129d4a299d428850496e19f9 (2.3.0 with a bit more commits)
 - Board & settings: STMicroelectronics:stm32:Nucleo_64:pnum=NUCLEO_WL55JC1,dbg=enable_log,opt=ogstd,upload_method=swdMethod

ABOSTM, 2022-08-11:
I am not so surprised as we mix Serial which make use of U(S)ART in Interrupt mode and core_debug which use U(S)ART in Polling mode. Not sure we tackled this competitive access to U(S)ART.

matthijskooijman, 2022-08-11:
Hm, would be good if this would work, since otherwise it is a lot harder for sketches to use the `core_debug()` logging features provided some libraries...

fpistm, 2022-08-11:
By default core debug use the same uart than Serial but it is possible to use an other one.

fpistm, 2022-09-08:
Hi @matthijskooijman 
I've made a PR with a fix. for this issue. Could you test it and give us a feedback, please?
Don't forget to remove your patch 😉

PR #1826 (2022-09-26): Core debug hardening
Fixes #1789 
Clean up the debug uart functions
Core debug is now half duplex by default if not linked to another Serial instance.
Tested:
- Serial and debug on the same U(S)ART
- Serial and debug on different U(S)ART
commit ff019c1b (2022-09-26): fix(uart): ensure debug uart is ready before send

Like state is ready not need to loop on the blocking
HAL_UART_Transmit(). If not ok it can be HAL_ERROR or HAL_TIMEOUT.
Moreover, it avoid to disable the U(S)ART IRQ which prevent to
receive data.

Fixes #1789
closed 2022-09-26 by fpistm