stm32duino/Arduino_Core_STM32 #1804: I2C: Wrong behavior with function setClock

camelator, 2022-08-15:
Considering 'M' a STM32Duino board as master.
Considering 'S' a STM32Duino board as slave.

1/ Having in setup()  for both M and S function TwoWire::setClock(100000) 
  => receiveEvent and requestEvent never called in S

2/ Having in setup() for M only function TwoWire::setClock(100000)
 => receiveEvent and requestEvent are called in S as expected.

Maybe the right way is to have this function working for masters only??? I don't know as I am not an i2c expert.

ABOSTM, 2022-08-26:
Hi @camelator,
The right way is to call setClock() function on both Master and Slave.
I tested on my side and there is this function on slave side, Interrupt are disabled when exiting this function (it should not).
Thanks for highlighting this issue.
I will submit a patch soon.

By the way if you simply wants to set clock @100000Hz,
then it is the default value, and you could remove both calls to setClock()

camelator, 2022-08-26:
Abostm,
many thanks for the answer,
I've noticed the default speed with a logic analyzer :)
I have to say I am working on having these 2 STM32 boards working as slave and master, and it looks like some issue or constraints on the slave side are driving errors on the master side. And the root cause is difficult to analyze....
Please have a look on the following issue:
https://github.com/stm32duino/Arduino_Core_STM32/issues/1805

I suspect the clock signal to be the root cause of the problems I am facing. It can explain why it works well with slave MCU and not with slave CPU. But definitely, I am not sure.

regards,

Christian

ABOSTM, 2022-08-29:
@camelator
Fix is available : #1818
Let me know if this solve your setClock issue.

PR #1818 (2022-08-30): fix(Wire): in case of slave, (re)enable interrupt when configuring clock
**Summary**
fix(Wire): in case of slave, (re)enable interrupt when configuring clock

Because TwoWire::setClock() calls i2c_setTiming()
which disable/(re)enable I2C Hardware,
it is necessary to restore interrupts.

Fixes #1804
closed 2022-08-30 by fpistm