stm32duino/Arduino_Core_STM32 #1809: trojan report

elgirsab, 2022-08-19:
Windows report a trojan in the:
Arduino_Core_STM32-main.zip file
the name of the trojan is : Trojan:Script/Wacatac.H!ml
in the details they indicate: This program is dangerous and executes commands from an attacker.
someone else experimented this issue?
what can we do??
thank you!

Elgir

fpistm, 2022-08-23:
Hi @elgirsab 

don't know why you download the main zip of this repo nor how. As in a standard way Arduino IDE or cli download the released packages. Anyway, there is no trojan, you can check the repo as it is fully open source and it mainly contains source code, no binary . It is used it daily on several machine with antivirus and Windows does not report any issue. Maybe a false positive.
I could not help on this.

closed 2022-08-23 by fpistm

elgirsab, 2022-08-25:
I noticed the problem occurs using the Auto Buld Marlin add on . now I did it just using Platformio and the problem was solved
thank you!
elgirsab

elgirsab, 2022-10-11:
Hello I found that the problem its with the Auto Marlin addon.   I just used the platformio and the problem disappeared.
thank you!
elgirsab