stm32duino/Arduino_Core_STM32 #1827: NUCLEO-U575ZI-Q memory size incorrect in Arduino IDE showing 256 kbytes should be 786 kbytes

Captainslarty, 2022-09-09:
Incorrect dynamic memory size being used by Arduino IDE for board Using NUCLEO-U575ZI-Q
IDE is showing - Maximum as 256 kbytes where device actually has **786 Kbyte**s of SRAM

Using NUCLEO-U575ZI-Q 
Arduino IDE 1.8.19
Core 2.3.0

example report from 'blink' sketch

Sketch uses 19820 bytes (0%) of program storage space. Maximum is 2097152 bytes.
Global variables use 1016 bytes (0%) of dynamic memory, leaving 261128 bytes for local variables. Maximum is 262144 bytes.

thank you
Joe

fpistm, 2022-09-09:
Hi  @Captainslarty 
Thanks to report this.
You're right, this mcu got up to 756Kb.
To define boards we generate several files automatically using the [STM32_open_pin_data](https://github.com/STMicroelectronics/STM32_open_pin_data) database. This is the same than the one used by STM32CubeMx tools.
The xml describing the STM32U575ZITxQ give 256Kb ram: https://github.com/STMicroelectronics/STM32_open_pin_data/blob/ae7fb660038b0bd9a2d50f0fa443dc024b15a4c9/mcu/STM32U575ZITxQ.xml#L13

Anyway the STM32CubeMx tools show the correct size. I have to investigate why/how and then probably update our scripts.
[image]

As a workaround you can edit the boards.txt and change the value:
https://github.com/stm32duino/Arduino_Core_STM32/blob/main/boards.txt#L200

Captainslarty, 2022-09-09:
Hi and thank you for the very prompt response. Much appreciated.
Please can I confirm with you the exact values that should be entered in these two lines ?

`Nucleo_144.menu.pnum.NUCLEO_U575ZI_Q.upload.maximum_size=2097152
Nucleo_144.menu.pnum.NUCLEO_U575ZI_Q.upload.maximum_data_size=262144`

fpistm, 2022-09-09:
This would be fine to see if all RAM are contiguous else there will be an issue. 😉 In a general way this core simply manage maximum contiguous RAM without ECC for more complex RAM configuration some extra config are required and custom linker script have to be used.

Captainslarty, 2022-09-09:
That is really appreciated. Fortunately or unfortunately ;) my level of knowledge is such that I will be highly unlikely to attempt more complex ram configurations :).  I have made the changes and all appears well.    
I know this next bit is going off topic slightly but please can I just enquire as to how I may get the NUCLEO-H7A3ZI-Q added to the Arduino core ? . I am not knowledgeable enough to produce a variant. Thanks again. Amazing support. Joe

fpistm, 2022-09-09:
I think it is OK anyway I would prefer warn you :wink:
About the Nucleo H7 it is already in the request variant list: https://github.com/stm32duino/Arduino_Core_STM32/issues/722
I've reference your comment too.

Captainslarty, 2022-09-09:
Thank you ever so much for all your kind help Frederic.
Regards,
Joe

fpistm, 2022-09-28:
Hi @Captainslarty 

issue confirmed internally and xml files will be corrected. I made a patch to fix them in the core.
About the size, We only set 768K as the SRAM4 is not consecutive:
[image]

Captainslarty, 2022-09-28:
Thank you again for you very hind help and response. Very much appreciated.
Joe

closed 2022-09-28 by fpistm

fpistm, 2022-11-30:
Hi @Captainslarty 
FYI, xml files have been updated in the latest tag of STM32_open_pin_data:
https://github.com/STMicroelectronics/STM32_open_pin_data/releases/tag/STM32CubeMX-DB.6.0.70

Ex: https://github.com/STMicroelectronics/STM32_open_pin_data/compare/STM32CubeMX-DB.6.0.60...STM32CubeMX-DB.6.0.70#diff-4fcffe7d2cbc0e786a7a7f6fe720ab7052aaa79194982bbbaada261c122a6df4L13

PR #1836 (2022-09-28): fix: U575Z(G-I)TxQ_U585ZITxQ ram issue
The xml describing those mcu give 256Kb ram.
Issue raised internally to fix this then it will be corrected in generated files.

Based on RM0456:
[image]

SRAM 1/2/3 are consecutive but SRAM4 not.

Fixes #1827
commit 5818d75a (2022-09-28): fix: U575Z(G-I)TxQ_U585ZITxQ ram issue

The xml describing those mcu give 256Kb ram.
Issue raised internally to fix this then it will be corrected
in generated files.

Fixes #1827