stm32duino/Arduino_Core_STM32 #1833: Wire add support for callback_function_t callbacks

hitech95, 2022-09-26:
**Is your feature request/improvement related to a problem? Please describe.**

The Wire methods onReceive and onRequest dont allow to pass a function via std::bind.
This would result to have to use a singleton to have access the hanlders if the I2C client code is an object like a library.

A solution similar to the attachInterrupt would solve the issue.

**Describe the solution you'd like**
Something like this (similar to what we do with `attachInterrupt()`)

```cpp
   _i2bus->begin(address + baseAddress);
   _i2bus->onReceive(std::bind(&I2CClient::onReceive, this));
   _i2bus->onRequest(std::bind(&I2CClient::onRequest, this));
```
My current workaround is to have multiple definitions of the static handling methods
each pair call the right object handling methods.

This is not the best solution with `callback_function_t` I could just pass the Wire object to class instanceand it will self manage.

fpistm, 2022-09-26:
Hi @hitech95 ,
Thanks for this issue.
Do not hesitate to provide a PR then we could review it carefully to ensure Arduino API compatibility :wink:

hitech95, 2022-09-26:
I'm testing a solution, for noew I have created a class that overrides the two methods above.
More tests are needed.

The code I'm working with:
```cpp
class MTwoWire : public TwoWire
{
private:
    std::function<void(int)> user_onReceive;
    std::function<void(void)> user_onRequest;

public:
    MTwoWire() : TwoWire(){};
    MTwoWire(uint32_t sda, uint32_t scl) : TwoWire(sda, scl){};

    void onReceive(void (*)(int));
    void onRequest(void (*)(void));

#ifdef __cplusplus
#include <functional>

    typedef std::function<void(int)> cb_function_receive_t;
    typedef std::function<void(void)> cb_function_request_t;

    void onReceive(cb_function_receive_t callback);
    void onRequest(cb_function_request_t callback);

#endif
};

void MTwoWire::onReceive(cb_function_receive_t function)
{
    user_onReceive = function;
}

void MTwoWire::onReceive(void (*function)(int))
{

    cb_function_receive_t _c = function;
    onReceive(_c);
}

// sets function called on slave read
void MTwoWire::onRequest(cb_function_request_t function)
{
    user_onRequest = function;
}

void MTwoWire::onRequest(void (*function)(void))
{
    cb_function_request_t _c = function;
    onRequest(_c);
}
```

hitech95, 2022-09-27:
@fpistm POC works, probably more tests are needed but so far so good.

Do you prefere a PR from/to main/master or you prefere a PR for a stable release?
Do you have a next branch to use as a base?

The patch is really simple, working out a draft PR right now.

fpistm, 2022-09-28:
PR will be merge in main. master is deprecated.

closed 2022-11-30 by fpistm