stm32duino/Arduino_Core_STM32 #1837: Problem accessing Serial3 on STM32F405RGT6 via suggested build_opt.h

simondlevy, 2022-09-29:
I have followed the [directions](https://github.com/stm32duino/wiki/wiki/API#hardwareserial)  for enabling Serial3  on my board, yet am getting the compiler error shown below.  The [PeripheralPins.c](https://github.com/stm32duino/Arduino_Core_STM32/blob/main/variants/STM32F4xx/F405RGT_F415RGT/PeripheralPins.c#L183-L198) for this board shows support for USART3, so I am not sure what I am doing wrong.  I've included my sketch, build_opt.h, and hardware settings as well.  As you can see from the sketch, my goal is to use serialEvent3() to read from this port.
[image]
[image]

Sketch:
```
static volatile uint32_t count;

void serialEvent3()
{
    while (Serial3.read()) {
        count++;
    }
}

void setup()
{
    Serial.begin(115200);

    Serial3.begin(115200);
}

void loop()
{
    // report the interrupt count twice per second

    static uint32_t prev;

    uint32_t time = millis();

    if (time-prev > 500) {
        Serial.println(count);
        prev = time;
    }
}
```

build_opt.h:
```
-DENABLE_HWSERIAL3
```

fpistm, 2022-09-29:
Hi @simondlevy 
there is a regression with 2.3.0 using the build_opt.h.

It is fixed in the main branch and will be available with the next release 2.4.0.
You could simply replace the prebuild.sh script by this file one: 
https://github.com/stm32duino/Arduino_Core_STM32/blob/main/system/extras/prebuild.sh

simondlevy, 2022-09-30:
Thanks, that worked!

closed 2022-09-30 by simondlevy