stm32duino/Arduino_Core_STM32 #1840: Generic H747XXX No compiling

ManuDLT, 2022-09-29:
I'm trying to use this core in a Portenta H7 Lite, that have a STM32H747XIH6. I suppose that I need to compile on the "Generic H747XXX" Board part number (I tried the 4 alternatives of 747), but when I try to compile a simple Blink example shows too many errors "detecting libraries included by: Arduino15\packages\STMicroelectronics\hardware\stm32\2.3.0\libraries\SrcWrapper\src....."

Here is a part of the error, I cut it because there are a lot of lines that says almost the same:
```
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Admin\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Admin\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Admin\Documents\Arduino\libraries -fqbn=STMicroelectronics:stm32:GenH7:pnum=GENERIC_H747IGTX,upload_method=swdMethod,xserial=disabled,usb=none,xusb=FS,opt=osstd,dbg=none,rtlib=nano -ide-version=10819 -build-path C:\Users\Admin\AppData\Local\Temp\arduino_build_761713 -warnings=none -build-cache C:\Users\Admin\AppData\Local\Temp\arduino_cache_378629 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.STM32Tools.path=C:\Users\Admin\AppData\Local\Arduino15\packages\STMicroelectronics\tools\STM32Tools\2.1.1 -prefs=runtime.tools.STM32Tools-2.1.1.path=C:\Users\Admin\AppData\Local\Arduino15\packages\STMicroelectronics\tools\STM32Tools\2.1.1 -prefs=runtime.tools.CMSIS.path=C:\Users\Admin\AppData\Local\Arduino15\packages\STMicroelectronics\tools\CMSIS\5.7.0 -prefs=runtime.tools.CMSIS-5.7.0.path=C:\Users\Admin\AppData\Local\Arduino15\packages\STMicroelectronics\tools\CMSIS\5.7.0 -prefs=runtime.tools.xpack-arm-none-eabi-gcc.path=C:\Users\Admin\AppData\Local\Arduino15\packages\STMicroelectronics\tools\xpack-arm-none-eabi-gcc\10.3.1-2.3 -prefs=runtime.tools.xpack-arm-none-eabi-gcc-10.3.1-2.3.path=C:\Users\Admin\AppData\Local\Arduino15\packages\STMicroelectronics\tools\xpack-arm-none-eabi-gcc\10.3.1-2.3 -verbose C:\Users\Admin\AppData\Local\Temp\arduino_modified_sketch_453822\Blink.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Admin\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Admin\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Admin\Documents\Arduino\libraries -fqbn=STMicroelectronics:stm32:GenH7:pnum=GENERIC_H747IGTX,upload_method=swdMethod,xserial=disabled,usb=none,xusb=FS,opt=osstd,dbg=none,rtlib=nano -ide-version=10819 -build-path C:\Users\Admin\AppData\Local\Temp\arduino_build_761713 -warnings=none -build-cache C:\Users\Admin\AppData\Local\Temp\arduino_cache_378629 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.STM32Tools.path=C:\Users\Admin\AppData\Local\Arduino15\packages\STMicroelectronics\tools\STM32Tools\2.1.1 -prefs=runtime.tools.STM32Tools-2.1.1.path=C:\Users\Admin\AppData\Local\Arduino15\packages\STMicroelectronics\tools\STM32Tools\2.1.1 -prefs=runtime.tools.CMSIS.path=C:\Users\Admin\AppData\Local\Arduino15\packages\STMicroelectronics\tools\CMSIS\5.7.0 -prefs=runtime.tools.CMSIS-5.7.0.path=C:\Users\Admin\AppData\Local\Arduino15\packages\STMicroelectronics\tools\CMSIS\5.7.0 -prefs=runtime.tools.xpack-arm-none-eabi-gcc.path=C:\Users\Admin\AppData\Local\Arduino15\packages\STMicroelectronics\tools\xpack-arm-none-eabi-gcc\10.3.1-2.3 -prefs=runtime.tools.xpack-arm-none-eabi-gcc-10.3.1-2.3.path=C:\Users\Admin\AppData\Local\Arduino15\packages\STMicroelectronics\tools\xpack-arm-none-eabi-gcc\10.3.1-2.3 -verbose C:\Users\Admin\AppData\Local\Temp\arduino_modified_sketch_453822\Blink.ino
Using board 'GenH7' from platform in folder: C:\Users\Admin\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.3.0
Using core 'arduino' from platform in folder: C:\Users\Admin\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.3.0
"C:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\tools\\STM32Tools\\2.1.1/win/busybox.exe" sh "C:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.3.0\\system/extras/prebuild.sh" "C:\\Users\\Admin\\AppData\\Local\\Temp\\arduino_build_761713" "C:\\Users\\Admin\\AppData\\Local\\Temp\\arduino_modified_sketch_453822" "C:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.3.0"
Detecting libraries used...
"C:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\tools\\xpack-arm-none-eabi-gcc\\10.3.1-2.3/bin/arm-none-eabi-g++" -mcpu=cortex-m7 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -DUSE_FULL_LL_DRIVER -mthumb "@C:\\Users\\Admin\\AppData\\Local\\Temp\\arduino_build_761713/sketch/build.opt" -c -Os -DNDEBUG -w -std=gnu++14 -ffunction-sections -fdata-sections -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -fno-use-cxa-atexit "-IC:\\Users\\Admin\\AppData\\Local\\Temp\\arduino_modified_sketch_453822" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.3.0\\cores\\arduino/avr" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.3.0\\cores\\arduino/stm32" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.3.0\\cores\\arduino/stm32/LL" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.3.0\\cores\\arduino/stm32/usb" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.3.0\\cores\\arduino/stm32/OpenAMP" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.3.0\\cores\\arduino/stm32/usb/hid" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.3.0\\cores\\arduino/stm32/usb/cdc" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.3.0\\system/Drivers/STM32H7xx_HAL_Driver/Inc" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.3.0\\system/Drivers/STM32H7xx_HAL_Driver/Src" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.3.0\\system/STM32H7xx" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.3.0\\system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.3.0\\system/Middlewares/ST/STM32_USB_Device_Library/Core/Src" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.3.0\\system/Middlewares/OpenAMP" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.3.0\\system/Middlewares/OpenAMP/open-amp/lib/include" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.3.0\\system/Middlewares/OpenAMP/libmetal/lib/include" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.3.0\\system/Middlewares/OpenAMP/virtual_driver" -w -x c++ -E -CC -DSTM32H7xx -DARDUINO=10819 -DARDUINO_GENERIC_H747IGTX -DARDUINO_ARCH_STM32 "-DBOARD_NAME=\"GENERIC_H747IGTX\"" "-DVARIANT_H=\"variant_generic.h\"" -DSTM32H747xx "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\tools\\CMSIS\\5.7.0/CMSIS/Core/Include/" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.3.0\\system/Drivers/CMSIS/Device/ST/STM32H7xx/Include/" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.3.0\\system/Drivers/CMSIS/Device/ST/STM32H7xx/Source/Templates/gcc/" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\tools\\CMSIS\\5.7.0/CMSIS/DSP/Include" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\tools\\CMSIS\\5.7.0/CMSIS/DSP/PrivateInclude" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.3.0\\cores\\arduino" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\ [... cut ...]
C:\Users\Admin\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.3.0\system/Drivers/CMSIS/Device/ST/STM32H7xx/Include/stm32h747xx.h:243:2: error: #error Please #define CORE_CM4 or CORE_CM7
  243 | #error Please #define CORE_CM4 or CORE_CM7
      |  ^~~~~
Using library SrcWrapper at version 1.0.1 in folder: C:\Users\Admin\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.3.0\libraries\SrcWrapper 
exit status 1
Error compiling for board Generic STM32H7 Series.

```
I already add that "#define CORE_CM7" but it doesn't work.

fpistm, 2022-09-29:
Hi
Could you try this patch: https://github.com/stm32duino/Arduino_Core_STM32/commit/94826835faee4cc35ac309bc820e2348bc1dd8ed

ManuDLT, 2022-09-29:
Yes, it compiles. Thank you.

Just to be sure, compile with any Generic H747XXX will works with the STM32H747XIH6 of the Portenta? 😅

fpistm, 2022-09-30:
If the portenta is a STM32H747XIH6  then this variant should be used:
https://github.com/stm32duino/Arduino_Core_STM32/tree/main/variants/STM32H7xx/H742X(G-I)H_H743X(G-I)H_H745X(G-I)H_H747X(G-I)H_H750XBH_H753XIH_H755XIH_H757XIH

but currently it is not declared, you can refer to this to see how to add it:
https://github.com/stm32duino/wiki/wiki/Add-a-new-variant-%28board%29

We have alread the request to add this board https://www.st.com/en/evaluation-tools/stm32h747i-disco.html based on the same MCU.