stm32duino/Arduino_Core_STM32 #1846: No definition for LPTIM in nucleo 32 Boards

VoitureRC, 2022-10-16:
**Describe the bug**
Hello ,
I'm don't see definition of LPTIM peripherals when i want to use these timers in Nucleo L432KC ,L031K6

And also in a STM32WLE55xx when i use these lines 
LPTIM_TypeDef *Instance = LPTIM1;
HardwareTimer *MyTim = new HardwareTimer(Instance); 

I got this error : 

```
C:\Users\Hamady\AppData\Local\Temp\.arduinoIDE-unsaved2022916-17548-11y6xbe.yfgl\sketch_oct16d\sketch_oct16d.ino:2:50: error: no matching function for call to 'HardwareTimer::HardwareTimer(LPTIM_TypeDef*&)'
    2 | HardwareTimer *MyTim = new HardwareTimer(Instance);
      |                                                  ^
In file included from C:\Users\Hamady\AppData\Local\Arduino15\packages\Move-X\hardware\stm32\1.0.0\cores\arduino/stm32/analog.h:45,
                 from C:\Users\Hamady\AppData\Local\Arduino15\packages\Move-X\hardware\stm32\1.0.0\cores\arduino/board.h:8,
                 from C:\Users\Hamady\AppData\Local\Arduino15\packages\Move-X\hardware\stm32\1.0.0\cores\arduino/wiring.h:40,
                 from C:\Users\Hamady\AppData\Local\Arduino15\packages\Move-X\hardware\stm32\1.0.0\cores\arduino/Arduino.h:36,
                 from C:\Users\Hamady\AppData\Local\Temp\arduino-sketch-25676C29BF517BF6658C6944726567F9\sketch\sketch_oct16d.ino.cpp:1:
C:\Users\Hamady\AppData\Local\Arduino15\packages\Move-X\hardware\stm32\1.0.0\cores\arduino/HardwareTimer.h:102:5: note: candidate: 'HardwareTimer::HardwareTimer(TIM_TypeDef*)'
  102 |     HardwareTimer(TIM_TypeDef *instance);
      |     ^~~~~~~~~~~~~
C:\Users\Hamady\AppData\Local\Arduino15\packages\Move-X\hardware\stm32\1.0.0\cores\arduino/HardwareTimer.h:102:32: note:   no known conversion for argument 1 from 'LPTIM_TypeDef*' to 'TIM_TypeDef*'
  102 |     HardwareTimer(TIM_TypeDef *instance);
      |                   ~~~~~~~~~~~~~^~~~~~~~
C:\Users\Hamady\AppData\Local\Arduino15\packages\Move-X\hardware\stm32\1.0.0\cores\arduino/HardwareTimer.h:101:5: note: candidate: 'HardwareTimer::HardwareTimer()'
  101 |     HardwareTimer();
      |     ^~~~~~~~~~~~~
C:\Users\Hamady\AppData\Local\Arduino15\packages\Move-X\hardware\stm32\1.0.0\cores\arduino/HardwareTimer.h:101:5: note:   candidate expects 0 arguments, 1 provided
C:\Users\Hamady\AppData\Local\Arduino15\packages\Move-X\hardware\stm32\1.0.0\cores\arduino/HardwareTimer.h:99:7: note: candidate: 'HardwareTimer::HardwareTimer(const HardwareTimer&)'
   99 | class HardwareTimer {
      |       ^~~~~~~~~~~~~
C:\Users\Hamady\AppData\Local\Arduino15\packages\Move-X\hardware\stm32\1.0.0\cores\arduino/HardwareTimer.h:99:7: note:   no known conversion for argument 1 from 'LPTIM_TypeDef*' to 'const HardwareTimer&'

exit status 1

Compilation error: no matching function for call to `'HardwareTimer::HardwareTimer(LPTIM_TypeDef*&)'`
```

**Desktop (please complete the following information):**
 - OS: WIN 11
 - Arduino IDE version: 2.0
 - STM32 core version: 2.3
 - Upload method: Mass storage

**Board (please complete the following information):**
 - Name: Nucleo L432KC ,L031K6

fpistm, 2022-10-17:
Hi @VoitureRC 
HardwareTimer does not support LPTIM. Only TIMx instance.
That's why you get this error.
If you want use LPTIM, then you will have to use HAL/LL API.
I guess it is kinked to this post:
https://www.stm32duino.com/viewtopic.php?p=10968#p10968

closed 2022-10-17 by fpistm

VoitureRC, 2022-10-18:
> Hi @VoitureRC HardwareTimer does not support LPTIM. Only TIMx instance. That's why you get this error. If you want use LPTIM, then you will have to use HAL/LL API. I guess it is kinked to this post: https://www.stm32duino.com/viewtopic.php?p=10968#p10968

Have you a exemple of LPTIM working with HAL  library (in stm32duino) beacause i struggle to acheive this even with copying code from the STM32 IDE

fpistm, 2022-10-18:
Done on the forum.