stm32duino/Arduino_Core_STM32 #1857: Reimplement SPI methods that use standard Arduino methods

Loud160, 2022-11-01:
What ever so called "improvements" have been made to the core regarding how SPI works should be removed or at the very least made backwards compatible to work with SPI libraries' designed for use the Arduino SPI format.  You can reference Roger Clarks Stmduino core for how SPI should be implemented so that it's compatible with Arduino SPI libraries.  If new functions were desired that are going to deviate from the standardized methods used by Arduino  that's fine, but they should NOT interfere with the core's default behavior when using SPI.

fpistm, 2022-11-01:
Hi.  Could you  be more precise. 
SPI library is compatible with Arduino API even if some extensions exist.
Some examples?

fpistm, 2022-11-06:
Closed as no feedback

closed 2022-11-06 by fpistm