stm32duino/Arduino_Core_STM32 #1858: Can not set the SCB->VTOR sucessfully?

KiraVerSace, 2022-11-02:
It still can't solve this problem after a few days of attempts.
I want to use IAP function in my code.
0x08000000   30K     for the bootloader
0x08007800   226K   for the user application
1. First I modified the ld file as follows:
2.

fpistm, 2022-11-02:
Which mcu?

KiraVerSace, 2022-11-03:
> Which mcu?

MCU: STM32L476RCT6
ENV: VSCODE+PLATFORMIO+Frame(Arduino)+MacOS

fpistm, 2022-11-03:
If `VECT_TAB_OFFSET` is set then it should be ok
https://github.com/stm32duino/Arduino_Core_STM32/blob/4e9dae5c87b5f5331545b02c0fa51d97228e4a37/system/STM32L4xx/system_stm32l4xx.c#L121-L124

and `SCB->VTOR` properly set:
https://github.com/stm32duino/Arduino_Core_STM32/blob/main/system/STM32L4xx/system_stm32l4xx.c#L191

I do not use PIO as we only support Arduino IDE. With Arduino IDE it works. So no more clue for you except check if PIO redefinition are properly take into account.

KiraVerSace, 2022-11-03:
> If `VECT_TAB_OFFSET` is set then it should be ok
>
> https://github.com/stm32duino/Arduino_Core_STM32/blob/4e9dae5c87b5f5331545b02c0fa51d97228e4a37/system/STM32L4xx/system_stm32l4xx.c#L121-L124
>
> and [...]

Thank you for your help, I find that the SCB->VTOR is unset after I download my application, as flow
[image]
it still 0x08000000, so I try debug my program, I step into the SystemInit,and I see SCB->VTOR is set to 0x08007800, and my program can run, but If I reset my mcu, it first run the bootloader, the 0xE000ED08(
SCB->VTOR ) will not change, i do not know why,
my program use the stm32duino freeRTOS.

KiraVerSace, 2022-11-03:
but as you see, I have set the offset in the ini file, and debug can succeed, but jump failed
[image]
I  can not figure out the reason, if the application complie in KEIL  it can succeed.

fpistm, 2022-11-03:
redefining `VECT_TAB_OFFSET` allows to redefine VTOR at the proper address when your application start but if your application does not start it is not set.
The upload offset is just to properly set where to flash and avoid write on your BL.

If keil succeed to start maybe it do some stuff to tell where to start. As stated when you flash 2 options:
1- board is reset (reboot) then BL have to jump at the correct @ 
2- the flash tools request to start at specific @ (jump) without reboot.

I could not help more on this as I have no more clue and relies to PIO.

closed 2022-11-03 by fpistm

KiraVerSace, 2022-11-03:
I had sloved the problem, as the link.
[PlatformIO Community](https://community.platformio.org/t/how-to-iap-in-stm32-correctly-there-is-some-problems/30276/4)

fpistm, 2022-11-03:
> I had sloved the problem, as the link. [PlatformIO Community](https://community.platformio.org/t/how-to-iap-in-stm32-correctly-there-is-some-problems/30276/4)

Thanks for the feedback.