stm32duino/Arduino_Core_STM32 #1860: STM32G0B1CBT unable to save to virtual EEPROM

alextrical, 2022-11-04:
**Describe the bug**
When using any example sketches that utilize the Virtual EEPROM no data is stored/retrievable

**To Reproduce**
`#include <EEPROM.h>

struct MyObject {
  float field1;
  byte field2;
  char name[10];
};

void setup() {

  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  float f = 123.456f;  //Variable to store in EEPROM.
  int eeAddress = 0;   //Location we want the data to be put.

  //One simple call, with the address first and the object second.
  EEPROM.put(eeAddress, f);

  Serial.println("Written float data type!");

  /** Put is designed for use with custom structures also. **/

  //Data to store.
  MyObject customVar = {
    3.14f,
    65,
    "Working!"
  };

  eeAddress += sizeof(float); //Move address to the next byte after float 'f'.

  EEPROM.put(eeAddress, customVar);
  Serial.println("Written custom data type! \n\n");

 //----------------------------------------------
  Serial.println("-------------------------");
  float g = 0.00f;   //Variable to store data read from EEPROM.
  Serial.print("Read float from EEPROM: ");

  //Get the float data from the EEPROM at position 'eeAddress'
  EEPROM.get(eeAddress, g);
  Serial.println(g, 3);    //This may print 'ovf, nan' if the data inside the EEPROM is not a valid float.
  secondTest(); //Run the next test.
}

struct MyObject2 {
  float field1;
  byte field2;
  char name[10];
};

void secondTest() {
  int eeAddress = sizeof(float); //Move address to the next byte after float 'f'.

  MyObject2 customVar2; //Variable to store custom object read from EEPROM.
  EEPROM.get(eeAddress, customVar2);

  Serial.println("Read custom object from EEPROM: ");
  Serial.println(customVar2.field1);
  Serial.println(customVar2.field2);
  Serial.println(customVar2.name);
}

void loop() {
  /* Empty loop */
}`

Steps to reproduce the behavior:
1. Flash sketch
2. View serial output
3. See error, all values returned as 0

**Expected behavior**
The sketch above stores and retrieves a value from Virtual EEPROM on a STM32F401 with no issue, but is unable to do so on a STM32G0B1

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04
 - Arduino IDE version: 1.8.19 & 2.0
 - STM32 core version: Main Branch (2.4.0?)
 - Tools menu settings if not the default:CDC Generic serial supersede USART
 - Upload method: DFU

**Board (please complete the following information):**
 - Name: BIG TREE TECH EBB42 V1.1 / GENERIC STM32G0B1CBT
 - Hardware Revision: V1.1

**Additional context**
I added support for this board to the repo a few months ago, but need some assistance to get the Virtual EEPROM support added. I have tried to define a 'FLASH_END' as '0x0801FFFFUL' (system/Drivers/CMSIS/Device/ST/STM32G0xx/Include/stm32g0b1xx.h:751) with no luck.

fpistm, 2022-11-08:
Hi @alextrical 
I've tested with a G01RB and it works. I did not have a G01CB. 
You should check if the Dual bank is enabled.
Anyway, I will not be able to help a lot on this. There already an issue on eeprom issue: #1500
EEPROM will requires some times to properly handle all possible use case.

alextrical, 2022-11-09:
Thank you for your response. It is rather odd. The G0B1CBT is a single bank device, but it may be configured as dual (I will see if i can figure out how to check that). I will hop over to that issue instead and see what i can do.

closed 2022-11-09 by alextrical