stm32duino/Arduino_Core_STM32 #1861: USB Keyboard not working and USB device not recognized on bluepill.

zyppx, 2022-11-08:
- Arduino_Core_STM32 version：2.3.0
- Arduino IDE 1.8.19
- STM32F103C8T6 BLUEPILL 

> I compiled the [Keyboard  example](https://github.com/stm32duino/Arduino_Core_STM32/blob/main/libraries/Keyboard/examples/KeyboardMessage/KeyboardMessage.ino), but USB device not recognized .
> 
> [image]
> 
> I can't find any help in issuse to make it work.
> 
> What should I do to make it work?

fpistm, 2022-11-08:
Hi @zyppx 
There are a lot of fake bluepill and even not a fake several has hardware issue with USB. You should search on the forum about this and get some advice to fix it.

closed 2022-11-08 by fpistm

JiriBilek, 2022-11-08:
First, check https://stm32duinoforum.com/forum/wiki_subdomain/index_title_Blue_Pill.html#Hardware_installation