stm32duino/Arduino_Core_STM32 #1879: [New Variant] -  L071

bilbo69, 2022-11-23:
Sirs,
you could evaluate the support the STM32L071 family?
In particular I should use the STM32L071CBT6 (48 pin/128K/ LQFP)

Thank you

fpistm, 2022-11-29:
Added to  [New Variant] request list #722

fpistm, 2022-11-29:
Duplicate of #1229

closed 2022-11-29 by fpistm

mamanbudiman, 2023-06-11:
Hello
I am using arduino IDE 2.1.0. I want to add a new variant for STM32L071RBT, but the MCU is still on the list of arduino IDE 2.1.0.
I am using this https://github.com/stm32duino/Arduino_Core_STM32/wiki/Add-a-new-variant-%28board%29 procedure to add a new variant.

Regards,

fpistm, 2023-06-12:
Hi @mamanbudiman 
There is an issue with Arduino IDE 2.x, the boards.txt is parsed when installed and so when you add a new board it is not parsed.
You have to delete some files of the Arduino IDE 2.
https://github.com/arduino/arduino-ide/issues/1030

Here the workaround:
https://github.com/arduino/arduino-ide/issues/1030#issuecomment-1152005617

mamanbudiman, 2023-06-12:
> Hi @mamanbudiman There is an issue with Arduino IDE 2.x, the boards.txt is parsed when installed and so when you add a new board it is not parsed. You have to delete some files of the Arduino IDE 2. [arduino/arduino-ide#1030](https://github.com/arduino/arduino-ide/issues/1030)
> 
> Here the workaround: [arduino/arduino-ide#1030 (comment)](https://github.com/arduino/arduino-ide/issues/1030#issuecomment-1152005617)

Ok, thanks @fpistm