stm32duino/Arduino_Core_STM32 #1883: TIMER behavior in STM32WLE

HamadyDIA, 2022-11-29:
Board : STM32WLE5JB

Software : [Move-Xduino](https://github.com/Move-X/Move-Xduino) . Based on stm32duino 2.3.0

Settings : Arduino 2.0.2 ? , optimise : smallest+LTO

-----------------------------------------------------------------------------------------------------------------

Hello,

I have an issue with the behavior of the timer with STM32WLE.

I need to have a timer that start every time a specific GPIO is high. 

After this i enable a interrupt to an other GPIO and count the numbers of risings eges for 100 ms (timer).

But all of this is managed with the chip on stop mode (mode 2 i think). 

And with my code sometimes the interrupt for the rising edge isn't working or the timer can't stop and the chip never goes in stop mode 

Here is my code 
`void CALLBACK_TIM2(){
  detachInterrupt(digitalPinToInterrupt(FREQUENCE_PIN)); 
  digitalWrite(LED_VERT,!digitalRead(LED_VERT)); 
  App_son.frequence_finale=App_son.frequence_int;
  tim2=true;
}

void INIT_TIMER_2(void) {
  TIM_2->setOverflow(10, HERTZ_FORMAT); // 100ms
  TIM_2->attachInterrupt(CALLBACK_TIM2);
  TIM_2->resume();
}

void FREQUENCE_2(){
  App_son.frequence_int++;
  digitalWrite(LED_BLEU,!digitalRead(LED_BLEU));   
}

void SINGAL_SON(void) {
  INIT_TIMER_2();
  TIM_2->setCount(0);
  TIM_2->attachInterrupt(CALLBACK_TIM2);
  TIM_2->resume();
  attachInterrupt(digitalPinToInterrupt(FREQUENCE_PIN), FREQUENCE_2, RISING);
  while(tim2==false){

  }
  
  tim2=false;
  //Serial2.begin(112500);
  //Serial2.println(App_son.frequence_finale);
  App_son.frequence_finale=0;
  App_son.frequence_int=0;
  //TIM_2->pause();
  //onLPEnter();  //Callback when the chip goes in stop mode 
}`
Thanks

HamadyDIA, 2022-11-29:
Thanks you for your message

(Question can we speak in french ?) 

> Can you tell me how you know timer doesn't stop?

I have the LED attached to the callback of the timer that continue to blink even without signal coming from outside.

>Can you tell me how you know chip never goes in stop mode ?

I have a PPK2 that mesure the current consumption and it stays at 6mA

>Also I don't see the code to go in stop mode (except a commented line).

With this fork made by move X the chip goes to sleep everytime it end a function. And is wakeup by the RTC.

`void CALLBACK_TIM2(){
  detachInterrupt(digitalPinToInterrupt(FREQUENCE_PIN)); 
  digitalWrite(LED_VERT,!digitalRead(LED_VERT)); 
  App_son.frequence_finale=App_son.frequence_int;
  tim2=true;
}

void INIT_TIMER_2(void) {
  TIM_2->setOverflow(10, HERTZ_FORMAT); // 100ms
  TIM_2->attachInterrupt(CALLBACK_TIM2);
  TIM_2->setCount(0);
  TIM_2->attachInterrupt(CALLBACK_TIM2);
  TIM_2->resume();
}

void FREQUENCE_2(){
  App_son.frequence_int++;
  digitalWrite(LED_BLEU,!digitalRead(LED_BLEU));   
}

void SINGAL_SON(void) {
  INIT_TIMER_2();

  attachInterrupt(digitalPinToInterrupt(FREQUENCE_PIN), FREQUENCE_2, RISING);
  while(tim2==false){
  }
  tim2=false;
  Serial2.begin(115200);
  Serial2.println(App_son.frequence_finale);
  App_son.frequence_finale=0;
  App_son.frequence_int=0;
}`

With these modification the timer IRQ and all IRQ work but the chip stays in run mode. And if i add TIM2->pause() after the while(tim2==false) the timer IRQ don't work. 

Thanks.

ABOSTM, 2022-11-29:
> With this fork made by move X the chip goes to sleep everytime it end a function 

Ok, I am not aware of moveX fork, but AFAICS, due to while loop in SINGAL_SON()
  ```
while(tim2==false){
  }
```
this prevent chip to go in lowpower mode (sleep ? or stop?) during the 100ms of timer2, because function doesn't end.
And system goes to LowPower when `SINGAL_SON()`  ends (after timer 2 elapse and while waiting for new trigger on 1st GPIO)

Also you initial talk about stop mode, but now you are talking about sleep mode. This is 2 different modes in STM32 chips, with different capabilities. Can you clarify ?

> And if i add TIM2->pause() after the while(tim2==false) the timer IRQ don't work.

Well, my guess would be that timer works for a while until pause() is reached.
Then maybe you miss to re enable your trigger  and thus restart the timer (by calling `SINGAL_SON()`) ?