stm32duino/Arduino_Core_STM32 #1885: Compile in Arduino IDE fails if username has spaces [2.3.0]

Sgw32, 2022-11-30:
**Describe the bug**

Compile in Arduino IDE fails if username has spaces
arm-none-eabi-g++: error: Team\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.3.0=.: No such file or directory

User name is "Test Team" (lol)
Forgot to change it :D

**To Reproduce**

Install Arduino Core from Boards Manager, Change the name to some with spaces, change board to Black Pill 411.
Compile sketch:
/*
  Blink onboard LED at 0.1 second interval
*/
void setup() {
  // initialize digital pin PB2 as an output.
  pinMode(PC13, OUTPUT); // LED connect to pin PC13
}
void loop() {
  digitalWrite(PC13, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(100);               // wait for 100mS
  digitalWrite(PC13, LOW);    // turn the LED off by making the voltage LOW
  delay(100);               // wait for 100mS
}

**Expected behavior**

Sketch compiles

**Screenshots**
https://i.ibb.co/kMhBMps/image.png

**Desktop (please complete the following information):**
Win10
Arduino IDE 1.8.13
STM32 core version: 2.3.0
Upload method: SWD

**Board (please complete the following information):**
Name: Black Pill 411

fpistm, 2022-11-30:
Hi @Sgw32 
It is fixed thanks https://github.com/stm32duino/Arduino_Core_STM32/pull/1759
Duplicate of #1806

fpistm, 2022-11-30:
As a workaround waiting the next release, simply replace the local prebuild.sh by this one:
https://github.com/stm32duino/Arduino_Core_STM32/blob/main/system/extras/prebuild.sh

closed 2022-11-30 by fpistm