stm32duino/Arduino_Core_STM32 #1887: Leafony STM32

masa6372, 2022-12-02:
Name: Leafony STM32
Document: https://docs.leafony.com/en/docs/leaf/processor/ap03/
Schematics: https://github.com/Leafony/HW-Design-Files/blob/master/AP03_STM32_MCU/AP03_STM32_MCU_Schematic.pdf
Where can be purchased: https://shop.leafony.com/en/collections/products/products/basic-kit-2

fpistm, 2022-12-05:
Related to https://www.stm32duino.com/viewtopic.php?t=1785

Added to  [New Variant] request list #722 
Feel free to contribute and submit PR to add it. 
To add a STM32 based board follows this [wiki](https://github.com/stm32duino/wiki/wiki/Add-a-new-variant-(board))

masa6372, 2022-12-06:
Hi, I'm wondering how to add a variant. Does it automatically created, or do we have to create it by following the wiki?

fpistm, 2022-12-06:
Some files already exists:
https://github.com/stm32duino/Arduino_Core_STM32/tree/main/variants/STM32L4xx/L452RC(I-T-Y)_L452RE(I-T-Y)x(P)_L462RE(I-T-Y)
You have to  add your own variant_*.* which describe your board then declare it to the boards.txt.
If you can start to define the pin mapping and I will try to help you to finish the variant.

masa6372, 2022-12-06:
Thanks！ 
I've added variant_LEAFONY_STM32.cpp and variant_LEAFONY_STM32.h. 
[https://github.com/Leafony/Arduino_Core_STM32/tree/main/variants/STM32L4xx/Leafony_STM32](url)

Can you tell where to modify?

fpistm, 2022-12-06:
@masa6372 
I've made the variant update:
https://github.com/stm32duino/Arduino_Core_STM32/pull/1889

Please, Could you review and test it?

masa6372, 2022-12-08:
LGTM!

commit 9adabec3 (2022-12-08): variants: add Leafony AP03

Fixes #1887
closed 2022-12-08 by fpistm