stm32duino/Arduino_Core_STM32 #1896: Preconfigured launch.json and other debug settings are needed for modern Arduino IDE 2.x

N0rbert, 2022-12-10:
**Is your feature request/improvement related to a problem? Please describe.**

Yes, it is related to debugging problem.  I'm unable to debug simple *Blink.ino* program on Nucleo F446RE board.

Steps to reproduce:

1. Download latest Arduino IDE 2.0.3, extract and launch it.
2. In Arduino IDE go to *File* → *Preferences* and set `https://github.com/stm32duino/BoardManagerFiles/raw/main/package_stmicroelectronics_index.json` as additional board URL for stm32duino.
1. In *Boards Manager* add *STM32 MCU based boards by STMicroelectronics*.
1. Select *Nucleo-64* board and set *Tools* → *Board part number* to *Nucleo F446RE*.
1. Set *Tools* → *Debug symbols and core logs* to *Symbols Enabled (-g)*.
1. Set *Tools* → *Optimize* to *Debug (-Og)*.
1. Try to debug the program.

[image]

Pressing *Debug → Start Debugging* open empty template for *launch.json* instead of normally preconfigured and reproducible file.

[image]

It is unclear what user should write in the fields on *launch.json* file 

```
{
  // Use IntelliSense to learn about possible attributes.
  // Hover to view descriptions of existing attributes.
  "version": "0.2.0",
  "configurations": [
  {
    "cwd": "${workspaceRoot}",
    "executable": "./bin/executable.elf",
    "name": "Debug with OpenOCD",
    "request": "launch",
    "type": "cortex-debug",
    "servertype": "openocd",
    "configFiles": [],
    "searchDir": [],
    "runToEntryPoint": "main",
    "showDevDebugOutput": "none"
  }
  ]
}
```

to get OpenOCD working with STM32.

**Describe the solution you'd like**

Arduino IDE 2.x detects preconfigured *launch.json*  for well known and mature stm32duino framework. 
Debugging is possible using OpenOCD software and embedded ST-Link hardware.

**Describe alternatives you've considered**

The debugging of STM32F446RE on Nucleo-64 F446RE is actually possible using VSCode with [*vscode-arduino*](https://marketplace.visualstudio.com/items?itemName=vsciot-vscode.vscode-arduino), [*cortex-debug*](https://marketplace.visualstudio.com/items?itemName=marus25.cortex-debug) and [*marus25.cortex-debug-dp-stm32f4*](https://marketplace.visualstudio.com/items?itemName=marus25.cortex-debug-dp-stm32f4) extensions by `openocd`  from Debian repository and `STM32F446x.svd` file from *marus25.cortex-debug-dp-stm32f4* extension.

**Additional context**

The current behavior was observed using latest Arduino IDE 
(Version: 2.0.3 ; Date: 2022-12-05T09:27:52.215Z ; CLI Version: 0.29.0 [76251df9] ) and stm32duino 2.3.0.

This issue was originally filed at https://github.com/arduino/arduino-ide/issues/1760 .

N0rbert, 2022-12-10:
I have just found this interesting thread  - https://www.stm32duino.com/viewtopic.php?f=18&p=10740 by @luca_stm32 .

I adapted this method to my Nucleo F446RE as follows on the same Debian 11 system:

```
sudo apt-get install openocd
sudo usermod -a -G dialout $USER
```

In Arduino IDE 2.0.3 at first I have installed *Arduino SAMD Boards (32-bits ARM Cortex-M0+)* from Board Manager.
Then I hacked needed files:

```
cp -v /usr/share/openocd/scripts/board/st_nucleo_f4.cfg ~/.arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0/variants/STM32F0xx/F030R8T/\{build.openocdscript\}
# ^^ note: I do not know why openocd needs above MCU and path! But this works!

cat <<EOF > /tmp/platform.txt.patch
--- /home/d/.arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0/platform.txt
+++ /home/d/.arduino15-hack/packages/STMicroelectronics/hardware/stm32/2.3.0/platform.txt
@@ -215,3 +215,42 @@
 tools.remoteproc_gen.upload.params.verbose=
 tools.remoteproc_gen.upload.params.quiet=
 tools.remoteproc_gen.upload.pattern="{busybox}" sh "{path}/{script}" generate "{build.path}/{build.project_name}.elf" "{build.path}/run_arduino_{build.project_name}.sh"
+
+# Debugger configuration (general options)
+# ----------------------------------------
+# EXPERIMENTAL feature:
+# - this is alpha and may be subject to change without notice
+debug.executable={build.path}/{build.project_name}.elf
+debug.toolchain=gcc
+debug.toolchain.path={runtime.tools.xpack-arm-none-eabi-gcc-10.3.1-2.3.path}/bin/
+debug.toolchain.prefix=arm-none-eabi-
+debug.server=openocd
+debug.server.openocd.path={runtime.tools.openocd-0.10.0-arduino7.path}/bin/openocd
+debug.server.openocd.scripts_dir={runtime.tools.openocd-0.10.0-arduino7.path}/share/openocd/scripts/
+debug.server.openocd.script={runtime.platform.path}/variants/{build.variant}/{build.openocdscript}
+
+#
+# OpenOCD sketch upload - version with configurable bootloader size
+# FIXME: this programmer is a workaround for default options being overwritten by uploadUsingPreferences
+#
+
+tools.openocd-withbootsize.path={runtime.tools.openocd-0.10.0-arduino7.path}
+tools.openocd-withbootsize.cmd=bin/openocd
+tools.openocd-withbootsize.cmd.windows=bin/openocd.exe
+
+tools.openocd-withbootsize.upload.params.verbose=-d2
+tools.openocd-withbootsize.upload.params.quiet=-d0
+tools.openocd-withbootsize.upload.pattern="{path}/{cmd}" {upload.verbose} -s "{path}/share/openocd/scripts/" -f "{runtime.platform.path}/variants/{build.variant}/{build.openocdscript}" -c "telnet_port disabled; program {{build.path}/{build.project_name}.bin} verify reset {bootloader.size}; shutdown"
+
+# Program flashes the binary at 0x0000, so use the linker script without_bootloader
+tools.openocd-withbootsize.program.params.verbose=-d2
+tools.openocd-withbootsize.program.params.quiet=-d0
+tools.openocd-withbootsize.program.pattern="{path}/{cmd}" {program.verbose} -s "{path}/share/openocd/scripts/" -f "{runtime.platform.path}/variants/{build.variant}/{build.openocdscript}" -c "telnet_port disabled; program {{build.path}/{build.project_name}.elf} verify reset; shutdown"
+
+tools.openocd-withbootsize.erase.params.verbose=-d3
+tools.openocd-withbootsize.erase.params.quiet=-d0
+tools.openocd-withbootsize.erase.pattern=
+
+tools.openocd-withbootsize.bootloader.params.verbose=-d2
+tools.openocd-withbootsize.bootloader.params.quiet=-d0
+tools.openocd-withbootsize.bootloader.pattern="{path}/{cmd}" {bootloader.verbose} -s "{path}/share/openocd/scripts/" -f "{runtime.platform.path}/variants/{build.variant}/{build.openocdscript}" -c "telnet_port disabled; init; halt; at91samd bootloader 0; program {{runtime.platform.path}/bootloaders/{bootloader.file}} verify reset; shutdown"

EOF

patch -p1 ~/.arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0/platform.txt < /tmp/platform.txt.patch
```

And as the result I can debug the program:

[image]

fpistm, 2022-12-12:
Hi @N0rbert 
We are aware of this and commented on the post you referred:
https://www.stm32duino.com/viewtopic.php?p=10740&sid=4664002bf596b3872fb0170d9f3d69a5#p10740
This will probably come.

fpistm, 2023-02-15:
Unfortunately, we do not succeed to reproduce the setup. My guess is the update of arduino-cli prevent to have it working.
Could you confirm it always work on your side with Arduino 2.0.3, please ?

fpistm, 2023-02-22:
We met some issues linked to Arduino IDE 2 which does not help to easily release that:
- [ ] https://github.com/arduino/arduino-ide/issues/1905
- [ ]

closed 2023-03-18 by fpistm

Phonog, 2025-12-02:
So, what is the status of including the debug feature in IDE V2 using STM32duino package 2.11?

fpistm, 2025-12-02:
> So, what is the status of including the debug feature in IDE V2 using STM32duino package 2.11?

Please avoid comment on old issue. 
As you can see it is closed as it is implemented in #1976 and had more enhancement later. Debug feature is supported .

PR #1976 (2023-03-18): Add debug openocd
**Summary**

This PR implements the debug feature in Arduino 2.0

* [x] Modifies variant script to generate openocd.cfg file for all variants
* [x] Adds debugger configuration to platform.txt
* [x] Update boards.txt for all families
* [x] Adds openocd.cfg for all variants

**Limitations**

Doesn't work if selected board is a Nucleo-* due to https://github.com/arduino/arduino-ide/issues/1960
Only use with Generic STM32xx series for the moment.

**Closing issues**

Fixes #1896