stm32duino/Arduino_Core_STM32 #1901: Program gets stuck, when set SCLK with PD1 on STM32G071

Eddddddddy, 2022-12-21:
**Describe the bug**
The problem is simple, I use STM32G071GBU6, using SPI to connect the sensor.
The IO configuration of SPI is as follows:
```
  SPI.setMISO(PA3);
  SPI.setMOSI(PA4);
  SPI.setSCLK(PD1);
  SPI.setSSEL(PD0);
```
This configuration is supported in STM32CUBEMX. And this IO configuration is also supported in the PeripheralPins.c file.
The problem I'm having is that as long as SCLK is configured as PD1, the program gets stuck when it executes to read the SPI. If I replace PD1 with other pins, it will not get stuck, and I can read and write SPI data normally.

**To Reproduce**
Using SPI example (DigitalPotControl.ino) 
```
// inslude the SPI library:
#include <SPI.h>

// set pin 10 as the slave select for the digital pot:
const int slaveSelectPin = 10;

void setup() {
  // set the slaveSelectPin as an output:
  Serial.begin(9600);
  pinMode(slaveSelectPin, OUTPUT);
  // initialize SPI:
  SPI.setMISO(PA3);
  SPI.setMOSI(PA4);
  //  SPI.setSCLK(PB8);
  SPI.setSCLK(PD1);
  SPI.setSSEL(PD0);
  SPI.begin();
}

void loop() {
  Serial.println("running");
  digitalPotWrite(0, 1);
}

void digitalPotWrite(int address, int value) {
  // take the SS pin low to select the chip:
  digitalWrite(slaveSelectPin, LOW);
  //  send in the address and value via SPI:
  SPI.transfer(address);
  SPI.transfer(value);
  // take the SS pin high to de-select the chip:
  digitalWrite(slaveSelectPin, HIGH);
}
```

**Expected behavior**
Serial will print 'running' without stopping with PD1.

**Desktop (please complete the following information):**
 - OS: macOS 13.1
 - Arduino IDE version: 1.8.19
 - STM32 core version: 2.3.0
 - Tools menu settings if not the default: 
[image]

 - Upload method: SWD

**Board (please complete the following information):**
 - Name: Custom G071

fpistm, 2022-12-21:
Hi @Eddddddddy 
Could you enable the Core logs in this menu:
[image]

and see if you get any message.

Eddddddddy, 2022-12-22:
Hi,
I enabled the Core logs, and no further information is output from the Serial port. And I tried Core Logs and Symbols Enabled (-g), the same result.

I'm not sure Core logs is output like this way.

fpistm, 2022-12-22:
By default, Core logs output on the same UART than Serial instance. It can also be defined to dedicated uart.

fpistm, 2022-12-22:
Anyway, it is possible there is no output. As you used custom board hard to tell what is the issue and config.

Eddddddddy, 2022-12-22:
thank you for your reply.
Although there is no official G071GBU6N board. I may need to buy a NUCLEO-G071RB to try and find the problem with the hardware.
But I think the problem might be with the software. No seriousness, just intuition. I'll probably keep digging for software issues while I wait for the NUCLEO to ship.

fpistm, 2022-12-22:
Schematics of the board could help.

Looking at your sketch there is an issue.
You defined the Hardware SPI SSEL and drive manually the CS pin I guess it conflicts.
Try to comment the setSSEL

```patch
// inslude the SPI library:
#include <SPI.h>

// set pin 10 as the slave select for the digital pot:
const int slaveSelectPin = 10;

void setup() {
  // set the slaveSelectPin as an output:
  Serial.begin(9600);
  pinMode(slaveSelectPin, OUTPUT);
  // initialize SPI:
  SPI.setMISO(PA3);
  SPI.setMOSI(PA4);
  SPI.setSCLK(PD1);
-   SPI.setSSEL(PD0);
+  // SPI.setSSEL(PD0);
  SPI.begin();
}

void loop() {
  Serial.println("running");
  digitalPotWrite(0, 1);
}

void digitalPotWrite(int address, int value) {
  // take the SS pin low to select the chip:
  digitalWrite(slaveSelectPin, LOW);
  //  send in the address and value via SPI:
  SPI.transfer(address);
  SPI.transfer(value);
  // take the SS pin high to de-select the chip:
  digitalWrite(slaveSelectPin, HIGH);
}

```

closed 2023-01-04 by fpistm