stm32duino/Arduino_Core_STM32 #1906: About the problem of software restart of STM32duino

CJT1111, 2022-12-23:
About the problem of software restart of STM32duino
I have two questions as follows:
1. When the stm32 accidentally crashes due to the single use of stm32, the stm32 will only be stuck there, while the 328p will automatically restart.
2. When I artificially want to restart stm32, generally speaking, I will write the following program on arduino. With this program, 328p will restart, while stm32 will be stuck there and will not restart.
Looking forward to your reply, I wish all the best!
`#include <Arduino.h>
void(* resetFunc) (void) = 0;//定义重启变量

void setup()
{
Serial.begin(115200);
Serial.println("启动成功");
delay(1000);
}

void loop()
{
Serial.println("重启");
delay(1000);
resetFunc();//重启
}`

fpistm, 2022-12-23:
Try 
`NVIC_SystemReset();`

CJT1111, 2022-12-23:
Thank you for your reply, your method is very effective! So if I want the program to be able to restart itself when it crashes, I need to write my own watchdog program?