stm32duino/Arduino_Core_STM32 #1925: Pin names messed up in STM32F105 and crash when writing digital pins > 35

CzokNorris, 2023-01-19:
I am using version stm32duino version 2.4.0
My Arduino IDE is 1.8.19
The chip is STM32F105RCT6 on a custom board

When I 
digitalWrite(36, HIGH);
or for example 
digitalWrite(PC_7, HIGH);
the program stops running on my board. 
And I have checked. The pin names are wrong. 

4 is supposed to be PA4, but in reality toggles PB5 (package pin 57)
5 is supposed to be PA5, but in reality toggles PB4 (package pin 56)
(supposed to be is taken from here: https://github.com/stm32duino/Arduino_Core_STM32/blob/main/variants/STM32F1xx/F105R(8-B-C)T/variant_generic.cpp)
I have not checked the other leds. 

This sketch stops running after 10 seconds (led stops blinking):
#define LED2 PB3

#define SimpleSwitchOutControl PC_7

void setup() {
  //LEDs: When driven LOW, they are off. When driven HIGH, they are illuminated
  pinMode(LED2, OUTPUT);
  pinMode(SimpleSwitchOutControl, OUTPUT);

  digitalWrite(LED2, LOW); 

}

void loop() {
  if (millis() % 1000 < 100){
    digitalWrite(LED2, HIGH);
  }else{
    digitalWrite(LED2, LOW);
  }

  if (millis() > 10000){
    if (millis() % 2000 < 1000){
      digitalWrite(SimpleSwitchOutControl, HIGH);
    }else{
      digitalWrite(SimpleSwitchOutControl, LOW);
    }
  }
  
}

What do do?

**Please, before reporting any issue**
- Make sure you are using the latest Arduino_Core_STM32 version.
  https://github.com/stm32duino/Arduino_Core_STM32/releases/latest
- Make sure the issue is not already reported/fixed on GitHub or discussed on the [stm32duino forum](https://stm32duino.com)
- Submit a GitHub issue only for reporting a problem related to the [Arduino_Core_STM32](https://github.com/stm32duino/Arduino_Core_STM32).
- **Avoid** to submit a GitHub issue for project troubleshooting.

Any questions/feedback/suggestions should be discussed on the [stm32duino forum](https://stm32duino.com):
  * questions on the [STM32 Core](https://www.stm32duino.com/viewforum.php?f=35)
  * bugs/enhancements on the [STM core: Bugs and enhancements](https://www.stm32duino.com/viewforum.php?f=38)

&#x26A0; When reporting any issue, please try to provide all relevant information to help on its resolution.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Complete source code which can be used to reproduce the issue. Please try to be as generic as possible (no extra code, extra hardware,...)

Steps to reproduce the behavior:
1. Plug
2. Press USER_BTN
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. Windows]
 - Arduino IDE version: [e.g. 1.8.8]
 - STM32 core version: [e.g. 1.4.0]
 - Tools menu settings if not the default: [e.g. Newlib Standard, No Serial]
 - Upload method: [e.g. SWD]

**Board (please complete the following information):**
 - Name: [e.g. Nucleo F429ZI]
 - Hardware Revision: [e.g. Rev B]
 - Extra hardware used if any: [e.g. X-Nucleo IKS01A1]

**Additional context**
Add any other context about the problem here.

CzokNorris, 2023-01-19:
[image]

CzokNorris, 2023-01-19:
Didnt see that Board part number can be edited. Leaving it here however, because the F105 is not in the list. Is there a way to add it to the list. Selected F103RCT from the list and now this error is not appearing

closed 2023-01-19 by CzokNorris

fpistm, 2023-01-20:
Hi,

First your sketch is not correct, you used PinName instead of pin number. So instead of `PC_7 `you have to use `PC7`.
To add a variant you can follow this wiki:
https://github.com/stm32duino/wiki/wiki/Add-a-new-variant-(board)

I've added the STM32F105RCT6 to the request list https://github.com/stm32duino/Arduino_Core_STM32/issues/722

brightproject, 2024-12-20:
> Hi,
> 
> First your sketch is not correct, you used PinName instead of pin number. So instead of `PC_7 `you have to use `PC7`. To add a variant you can follow this wiki: https://github.com/stm32duino/wiki/wiki/Add-a-new-variant-(board)
> 
> I've added the STM32F105RCT6 to the request list #722

New link on `wiki` @fpistm 

> https://github.com/stm32duino/Arduino_Core_STM32/wiki/Add-a-new-variant-(board)