stm32duino/Arduino_Core_STM32 #1940: Cant Do Serial and LoRaWAN on STM32WLE5JC (Seeed LoRa E5 Mini)

rajSilikon, 2023-02-03:
Having Issues with Seeed Lora-e5 mini

STM32WLE5JC

I tried blinking an LED and fade an LED just to check if it communicates with the chip.

Blink LED and Fade LED works fine no issues.

but I cant do other than that, for example Serial Read / Write not working.
LoRaWAN code also not working.

[image]

Am I missing something or is there a bug?

fpistm, 2023-02-03:
Hi @rajSilikon 
by default, generic WLE5JC defined default Serial instance to be on:
https://github.com/stm32duino/Arduino_Core_STM32/blob/21dce35952408002a1c6770db406b073339a0bd4/variants/STM32WLxx/WL54JCI_WL55JCI_WLE4J(8-B-C)I_WLE5J(8-B-C)I/variant_generic.h#L126-L138

But based on this pinout:
https://files.seeedstudio.com/products/317990687/image/3001615286723_.pic_hd.jpg

It seems it should be PB6/PB7.
Before
`Serial.begin(speed);`
Add:
```
Serial.setRx(PB7);
Serial.setTx(PB6);
```

rajSilikon, 2023-02-06:
I was able to get serial to work using 
Serial.setRx(PB7);
Serial.setTx(PB6); 

Is LoRaWAN in Arduino IDE supported on this chip already? 

Cant seem to get that to work

fpistm, 2023-02-06:
Fine.
For LoRaWan, yes it could be supported thanks:
https://github.com/stm32duino/STM32LoRaWAN

Anyway, some config need to be updated. We will update the library to be able to redefine easily the rfswitch pins.
Like it was done for the RAK:
https://github.com/stm32duino/STM32LoRaWAN/issues/7

closed 2023-02-06 by fpistm

fpistm, 2023-04-07:
Fixed by #1997