stm32duino/Arduino_Core_STM32 #1950: Missing pins definitions for buttons and LEDs on P-NUCLEO-WB55RG ?

Yannick-Marietti, 2023-02-15:
Hello, 

Core : 2.4.0
Arduino IDE : 2.0.3

I am trying to reproduce the blink example (with also a polling on user buttons)  on a NUCLEO_WB55RG board.

As mentionned in variant_generic.c file for WB55R(C-E-G)V variant, LED_BUILTIN and USER_BTN are both #defined as PNUM_NOT_DEFINED.  As a consequence, the blink.ino example is not working on this board.

So I started a new project on P-NUCLEO_WB55RG  in STm32CubeIDE  to get the correct pins alisases and put them inside my STM32duino sketch :

#define LED1 PB5
#define LED2 PB0
#define LED3 PB1
#define SW1 PC4
#define SW2 PD0
#define SW3 PD1

No errors occurs while compiling.
But substituting  those inside blink.ino (for LEDs, for example, but issue is the same for buttons) just doesn't work.
Is it possible that some definitions are missing inside the STM32duino core board package ?

Best regards

Yannick

fpistm, 2023-02-15:
Hi @Yannick-Marietti 
Generic is for generic mcu, so this is normal it was defined to not defined.
For dedicated board like the P Nucleo WB55RG then you have to look to the correct variant_*.[ch] files where you will find the pins definitions of the baord:
https://github.com/stm32duino/Arduino_Core_STM32/blob/2e03c807f9d568b13438abd8ef57bb531b04075a/variants/STM32WBxx/WB55R(C-E-G)V/variant_P_NUCLEO_WB55RG.h#L82-L100

closed 2023-02-15 by fpistm

Yannick-Marietti, 2023-02-15:
Thanks
I had a look in the wrong definition file !

I confirm it's finally  working, but there was also something unclear with Arduino IDE 2 ; I had to restart it to make blink working on my Nucleo WB55 board (?).

BR

Yannick