stm32duino/Arduino_Core_STM32 #1951: The problem of Stm32G030F6Px Serial in Generic STM32G0 series.

13507086298, 2023-02-16:
The problem of Stm32G030F6Px Serial in Generic STM32G0 series.

I am using stm32duino 2.4.0, arduino 1.8.19 version, when using the hardware serial of USART1, there is no output. Until I use setRx to reset the port, the output is normal.

I guess there is some glitch with the port mapping in the board package?

My English is not good, thank you for understanding

`
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  Serial.setRx(PB7);    //必须同时使用方法设定 RX->PB7，TX->PB6，否则Serial没有输出。
  Serial.setTx(PB6);    //通过查数据表得知，Usart1_TX是pin PB6，虽然pin PB6和PB3都是用的pin20脚，但需要单独引用
                        //而HardwareSerial类中定义的Serial，实际引用的是PB3~PB5中的一个。所以不setTx为PB6，串口是不输出的
                        
  Serial.begin(115200);

  pinMode(PA4, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(PA4, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(PA4, LOW);    // turn the LED off by making the voltage LOW
 Serial.println("Stm32G030F6P6: Hello,world!");
  delay(1000);                       // wait for a second
}`

fpistm, 2023-02-16:
I see no issue.
By default, Serial instance is mapped to Serial2 which uses pins PA3/PA2:
https://github.com/stm32duino/Arduino_Core_STM32/blob/2e03c807f9d568b13438abd8ef57bb531b04075a/variants/STM32G0xx/G030F6P/variant_generic.h#L111-L123

So if you want use PB7/PB6 you do the right thing.

closed 2023-02-16 by fpistm

13507086298, 2023-02-16:
Thanks, indeed, as you said, Serial is redirected to Serial2.
So if I need to use 2 Serials, I can declare another hardware serial port. I always thought that serial corresponds to USART1, and Serial2 corresponds to USART2.
But what I can't understand is why Serial will be remapped to Serial2
Thanks, indeed, as you said, Serial is redirected to Serial2.
So if I need to use 2 Serials, I can declare another hardware serial port. I always thought that serial corresponds to USART1, and Serial2 corresponds to USART2.
But what I can't understand is why Serial will be remapped to Serial2

fpistm, 2023-02-16:
Well, Serial is simply an alias to a Serialx where x is the instance. By default, only one instance are created. Generic variant are automatically generated and so the first USART pin available are used as the default. In this case PA3 and PA2 which relies to USART2. That's all. If you need a second one then simply instantiate it as described in the wiki.
https://github.com/stm32duino/wiki/wiki/API#hardwareserial

13507086298, 2023-02-16:
finally understood. thanks for your explanation.