stm32duino/Arduino_Core_STM32 #1955: Commit 45c10e61 introduces bug?

honnet, 2023-02-22:
Before commit 45c10e61 my compilations work well, but after that, I get this error:

`fork/exec {runtime.tools.xpack-arm-none-eabi-gcc-12.2.1-1.2.path}/bin/arm-none-eabi-g++: no such file or directory`

My arm-none-eabi-g++ was installed automatically by the Arduino Board Manager, and it seems to be the problem.

=>  Is there an official procedure to update it? (recursive git pull?)

fpistm, 2023-02-22:
That is normal as the toolchain version has changed in the main branch. 
You have to install it manually.
A dedicated package index json file with links to the new toolchain is available on the dev branch:
https://github.com/stm32duino/BoardManagerFiles/blob/dev/package_stmicroelectronics_index.json

closed 2023-02-22 by fpistm

honnet, 2023-02-22:
Thanks Frédéric !

For those who would happen to have the same problem as me, this is the link that allowed getting the expected toolchain (linux 64bit):
https://github.com/xpack-dev-tools/arm-none-eabi-gcc-xpack/releases/download/v12.2.1-1.2/xpack-arm-none-eabi-gcc-12.2.1-1.2-linux-x64.tar.gz

Source:
https://github.com/stm32duino/BoardManagerFiles/blob/dev/package_stmicroelectronics_index.json#L577

SFE-Brudnerd, 2023-02-27:
Hi @fpistm thank you for talking to this issue, however I don't agree that this issue should be labelled Invalid. While the BoardManagerFiles repo has dev branch, this repo does not. Therefore, a breaking change was added to the main branch of this repo. Anyone that was working with this repo expecting it to be functional are now having to find this specific issue and manually solve this. IMO, there should be a dev branch for this repo as well. Anyone new to working on their own new variant and having only downloaded this repo direct from GH might not realize that they have a broken development version, nor would they necessarily have the correct boards file json downloaded since it lives in a separate repo that doesn't track changes across back to this repo.

fpistm, 2023-02-28:
Hi @SFE-Brudnerd 
> While the BoardManagerFiles repo has dev branch, this repo does not.

The dev branch of the BoardManagerFiles  is mainly to provide tools dependencies for the CI. No code are stored on this repo.

For this repo, no reason to create a dev branch, while no tag is created the main contains all new features and fixes. the next release will simply have a new toolchain and the way Arduino handle tools dependencies requires to set the version in platform.txt. So yes you have to install the new toolchain or change the version in the platform.txt. When you used the git repository you are in a development environment and so it can have so dependencies to update.
Creating a dev branch would simply implies to merege all in this branch then merge in main branch anyway, the problem will be the same and you will have to install the dependencies. As all PR have to be rebase on top of the main (or dev branch as you suggest) then no benefits.

SFE-Brudnerd, 2023-02-28:
You make a good point! Thanks for updating the wiki. I think that's a good compromise.