stm32duino/Arduino_Core_STM32 #1956: rosserial_arduino for stm32

nopnop2002, 2023-02-23:
**Describe the solution you'd like**

I want rosserial_arduino_lib available with this core library

fpistm, 2023-02-23:
Hi,
I guess this is this lib:
https://github.com/frankjoshua/rosserial_arduino_lib

If the lib uses standard arduino API it should works else you will have to modify it or request to their repo to support stm32  core.
Here it is for core issue.

closed 2023-02-23 by fpistm

nopnop2002, 2023-02-23:
The library you pointed to, which is detailed here, may depend on rogerclarkmelbourne's core.

http://wiki.ros.org/rosserial_arduino

nopnop2002, 2023-02-23:
Adding this line made it work.

Serial1 is used as the serial object.

TX=PA9/RX=PA10 for BluePill and BlackPill.

```
Adding this line made it work.
```