stm32duino/Arduino_Core_STM32 #1973: erorr compile srcwrapper

Desember5, 2023-03-09:
how to fix??

Arduino: 1.8.19 (Windows 7), Board: "Generic STM32F1 series, BluePill F103C8, STM32CubeProgrammer (SWD), Enabled (generic 'Serial'), None, Low/Full Speed, Smallest (-Os default), None, Newlib Nano (default)"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\AWR\AppData\Local\Arduino15\packages -hardware C:\Users\AWR\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\AWR\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\AWR\Documents\Arduino\libraries -fqbn=STMicroelectronics:stm32:GenF1:pnum=BLUEPILL_F103C8,upload_method=swdMethod,xserial=generic,usb=none,xusb=FS,opt=osstd,dbg=none,rtlib=nano -ide-version=10819 -build-path C:\Users\AWR\AppData\Local\Temp\arduino_build_887225 -warnings=none -build-cache C:\Users\AWR\AppData\Local\Temp\arduino_cache_347857 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xpack-arm-none-eabi-gcc.path=C:\Users\AWR\AppData\Local\Arduino15\packages\STMicroelectronics\tools\xpack-arm-none-eabi-gcc\10.3.1-2.3 -prefs=runtime.tools.xpack-arm-none-eabi-gcc-10.3.1-2.3.path=C:\Users\AWR\AppData\Local\Arduino15\packages\STMicroelectronics\tools\xpack-arm-none-eabi-gcc\10.3.1-2.3 -prefs=runtime.tools.CMSIS.path=C:\Users\AWR\AppData\Local\Arduino15\packages\STMicroelectronics\tools\CMSIS\5.7.0 -prefs=runtime.tools.CMSIS-5.7.0.path=C:\Users\AWR\AppData\Local\Arduino15\packages\STMicroelectronics\tools\CMSIS\5.7.0 -prefs=runtime.tools.STM32Tools.path=C:\Users\AWR\AppData\Local\Arduino15\packages\STMicroelectronics\tools\STM32Tools\2.1.1 -prefs=runtime.tools.STM32Tools-2.1.1.path=C:\Users\AWR\AppData\Local\Arduino15\packages\STMicroelectronics\tools\STM32Tools\2.1.1 -verbose C:\Users\AWR\AppData\Local\Temp\untitled1034123064.tmp\sketch_mar09a\sketch_mar09a.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\AWR\AppData\Local\Arduino15\packages -hardware C:\Users\AWR\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\AWR\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\AWR\Documents\Arduino\libraries -fqbn=STMicroelectronics:stm32:GenF1:pnum=BLUEPILL_F103C8,upload_method=swdMethod,xserial=generic,usb=none,xusb=FS,opt=osstd,dbg=none,rtlib=nano -ide-version=10819 -build-path C:\Users\AWR\AppData\Local\Temp\arduino_build_887225 -warnings=none -build-cache C:\Users\AWR\AppData\Local\Temp\arduino_cache_347857 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xpack-arm-none-eabi-gcc.path=C:\Users\AWR\AppData\Local\Arduino15\packages\STMicroelectronics\tools\xpack-arm-none-eabi-gcc\10.3.1-2.3 -prefs=runtime.tools.xpack-arm-none-eabi-gcc-10.3.1-2.3.path=C:\Users\AWR\AppData\Local\Arduino15\packages\STMicroelectronics\tools\xpack-arm-none-eabi-gcc\10.3.1-2.3 -prefs=runtime.tools.CMSIS.path=C:\Users\AWR\AppData\Local\Arduino15\packages\STMicroelectronics\tools\CMSIS\5.7.0 -prefs=runtime.tools.CMSIS-5.7.0.path=C:\Users\AWR\AppData\Local\Arduino15\packages\STMicroelectronics\tools\CMSIS\5.7.0 -prefs=runtime.tools.STM32Tools.path=C:\Users\AWR\AppData\Local\Arduino15\packages\STMicroelectronics\tools\STM32Tools\2.1.1 -prefs=runtime.tools.STM32Tools-2.1.1.path=C:\Users\AWR\AppData\Local\Arduino15\packages\STMicroelectronics\tools\STM32Tools\2.1.1 -verbose C:\Users\AWR\AppData\Local\Temp\untitled1034123064.tmp\sketch_mar09a\sketch_mar09a.ino

Using board 'GenF1' from platform in folder: C:\Users\AWR\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.4.0

Using core 'arduino' from platform in folder: C:\Users\AWR\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.4.0

"C:\\Users\\AWR\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\tools\\STM32Tools\\2.1.1/win/busybox.exe" sh "C:\\Users\\AWR\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.4.0\\system/extras/prebuild.sh" "C:\\Users\\AWR\\AppData\\Local\\Temp\\arduino_build_887225" "C:\\Users\\AWR\\AppData\\Local\\Temp\\untitled1034123064.tmp\\sketch_mar09a" "C:\\Users\\AWR\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.4.0"

Detecting libraries used...

"C:\\Users\\AWR\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\tools\\xpack-arm-none-eabi-gcc\\10.3.1-2.3/bin/arm-none-eabi-g++" -mcpu=cortex-m3 -DUSE_FULL_LL_DRIVER -mthumb "@C:\\Users\\AWR\\AppData\\Local\\Temp\\arduino_build_887225/sketch/build.opt" -c -Os -DNDEBUG -w -std=gnu++14 -ffunction-sections -fdata-sections -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -fno-use-cxa-atexit "-IC:\\Users\\AWR\\AppData\\Local\\Temp\\untitled1034123064.tmp\\sketch_mar09a" "-IC:\\Users\\AWR\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.4.0\\cores\\arduino/avr" "-IC:\\Users\\AWR\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.4.0\\cores\\arduino/stm32" "-IC:\\Users\\AWR\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.4.0\\cores\\arduino/stm32/LL" "-IC:\\Users\\AWR\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.4.0\\cores\\arduino/stm32/usb" "-IC:\\Users\\AWR\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.4.0\\cores\\arduino/stm32/OpenAMP" "-IC:\\Users\\AWR\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.4.0\\cores\\arduino/stm32/usb/hid" "-IC:\\Users\\AWR\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.4.0\\cores\\arduino/stm32/usb/cdc" "-IC:\\Users\\AWR\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.4.0\\system/Drivers/STM32F1xx_HAL_Driver/Inc" "-IC:\\Users\\AWR\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.4.0\\system/Drivers/STM32F1xx_HAL_Driver/Src" "-IC:\\Users\\AWR\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.4.0\\system/STM32F1xx" "-IC:\\Users\\AWR\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.4.0\\system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc" "-IC:\\Users\\AWR\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.4.0\\system/Middlewares/ST/STM32_USB_Device_Library/Core/Src" "-IC:\\Users\\AWR\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.4.0\\system/Middlewares/OpenAMP" "-IC:\\Users\\AWR\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.4.0\\system/Middlewares/OpenAMP/open-amp/lib/include" "-IC:\\Users\\AWR\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.4.0\\system/Middlewares/OpenAMP/libmetal/lib/include" "-IC:\\Users\\AWR\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.4.0\\system/Middlewares/OpenAMP/virtual_driver" -w -x c++ -E -CC -DSTM32F1xx -DARDUINO=10819 -DARDUINO_BLUEPILL_F103C8 -DARDUINO_ARCH_STM32 "-DBOARD_NAME=\"BLUEPILL_F103C8\"" "-DVARIANT_H=\"variant_PILL_F103Cx.h\"" -DSTM32F103xB -DHAL_UART_MODULE_ENABLED "-IC:\\Users\\AWR\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\tools\\CMSIS\\5.7.0/CMSIS/Core/Include/" "-IC:\\Users\\AWR\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.4.0\\system/Drivers/CMSIS/Device/ST/STM32F1xx/Include/" "-IC:\\Users\\AWR\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.4.0\\system/Drivers/CMSIS/Device/ST/STM32F1xx/Source/Templates/gcc/" "-IC:\\Users\\AWR\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\tools\\CMSIS\\5.7.0/CMSIS/DSP/Include" "-IC:\\Users\\AWR\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\tools\\CMSIS\\5.7.0/CMSIS/DSP/PrivateInclude" "-IC:\\Users\\AWR\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.4.0\\cores\\arduino" "-IC:\\Users\\AWR\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.4.0\\variants\\STM32F1xx\\F103C8T_F103CB(T-U)" "C:\\Users\\AWR\\AppData\\Local\\Temp\\arduino_build_887225\\sketch\\sketch_mar09a.ino.cpp" -o nul

[... cut ...] l\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.4.0\\cores\\arduino" "-IC:\\Users\\AWR\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.4.0\\variants\\STM32F1xx\\F103C8T_F103CB(T-U)" "C:\\Users\\AWR\\AppData\\Local\\Temp\\arduino_build_887225\\sketch\\sketch_mar09a.ino.cpp" -o "C:\\Users\\AWR\\AppData\\Local\\Temp\\arduino_build_887225\\preproc\\ctags_target_for_gcc_minus_e.cpp"

exit status -1073741819

Error compiling for board Generic STM32F1 series.

fpistm, 2023-03-09:
Hi @Desember5
Don't know what is the issue. Seems more an Arduino ide issue.
Build works as expected for me. Maybe path issue due to windows limitations on path higher than 256.
Or maybe issue on win7. Unfortunately I could not try.

Desember5, 2023-03-09:
I tried to compile the blink led example code on the stm32F1 board using the stm32duino board but during the compilation process there is a message error while detecting libraries included by c:\users\awr\appdata\local\temp\arduino_build_997423\sketch\srcwrapper.cpp even though when I test compile the same code on another board I don't find an error like that... my question is how to solve this

fpistm, 2023-03-09:
Try to reinstall.

Desember5, 2023-03-09:
I have tried deleting and reinstalling the board but the error message is still the same... I even tried using Windows 10 and the results are still the same

fpistm, 2023-03-10:
I've made some search about this error and found several entries in Arduino community and several refer to this issue:
https://github.com/arduino/Arduino/issues/8394#issuecomment-454693547

closed 2023-03-10 by fpistm