stm32duino/Arduino_Core_STM32 #1978: H5 Support?

runger1101001, 2023-03-15:
**Is your feature request/improvement related to a problem? Please describe.**

The new H5 series is not yet included in the generic variants. I got a Nucleo-64 STM32H503 here in front of me from embeddedWorld '23, and can't program it using Arduino :-(

**Describe the solution you'd like**

It would be great if generic H5 variants were added, and even better if you could add the Nucleos as well :-)

fpistm, 2023-03-17:
Hi,
Currently, no H5 release planned at the moment, but the request is taken into account.

runger1101001, 2023-03-17:
Boohoo, well, looking forward to it...

benjaminnestler, 2023-03-28:
I've also this Nucleo-H563ZI development board ... so the integration would be very nice ;-)

dojyorin, 2023-04-05:
I'm also very interested in the STM32H563/573.
(mouser has no stock😭)
I'm looking forward to be able to use it as an Arduino😆

dojyorin, 2023-05-20:
@fpistm We STM board users are very grateful for your great contribution.
I would like to return the favor even just a little, and would like to donate to you using GitHub Sponsers.
(That said, I can only donate about 1 cup of coffee each month...)
This is called "nagesen" in our Japanese, and refers to the act of donating to those who have made great contribution in gratitude and wishing for their future success.

closed 2023-05-26 by fpistm

F14V, 2024-08-07:
Could support for STM32H503CB be added? WeActStudio has released a pretty nice [board ](https://github.com/WeActStudio/WeActStudio.STM32H503CoreBoard)using that chip and probably there are more to come with H5 MCUs.