stm32duino/Arduino_Core_STM32 #1980: Serial MOnitor not working on Stm32F103R

zlomennypez, 2023-03-20:
Hello,

If

**Please, before reporting any issue**
- Make sure you are using the latest Arduino_Core_STM32 version.
  https://github.com/stm32duino/Arduino_Core_STM32/releases/latest
- Make sure the issue is not already reported/fixed on GitHub or discussed on the [stm32duino forum](https://stm32duino.com)
- Submit a GitHub issue only for reporting a problem related to the [Arduino_Core_STM32](https://github.com/stm32duino/Arduino_Core_STM32).
- **Avoid** to submit a GitHub issue for project troubleshooting.

Any questions/feedback/suggestions should be discussed on the [stm32duino forum](https://stm32duino.com):
  * questions on the [STM32 Core](https://www.stm32duino.com/viewforum.php?f=35)
  * bugs/enhancements on the [STM core: Bugs and enhancements](https://www.stm32duino.com/viewforum.php?f=38)

&#x26A0; When reporting any issue, please try to provide all relevant information to help on its resolution.

**Describe the bug**
Serial Monitor not working, if I choose any od STM32F103R board. If I choose in Adruino IDE Stm32F103C or Stm32F103Z, it works, but I dont think, It´s good solution, if the board is STM32F103R.

**To Reproduce**
Any Code

Steps to reproduce the behavior:
1. Plug
2. Press USER_BTN
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 -Windows
 - 2.0.4
 - STM32 core version: Dont know - latest
 - Tools menu settings if not the default: [e.g. Newlib Standard, No Serial]
 - Upload method: [e.g. SWD]

**Board (please complete the following information):**
 - Stm32F103RET6

**Additional context**
Add any other context about the problem here.

zlomennypez, 2023-03-20:
> You probably do not use the usart pins you think. Default pins are defined here:
> 
> https://github.com/stm32duino/Arduino_Core_STM32/blob/5c87586d8cbefe314ccd74297e919c045335619c/variants/STM32F1xx/F103R(C-D-E)T/variant_generic.h#L177-L184

Thank you, but PA2 PA3 are TX2 RX2 pins,  TX1 is PB6 and RX1 is PB7. It the same as for all f103 board. Onhttps://www.st.com/resource/en/datasheet/stm32f103ve.pdf  page 36.

And if I change the board in Arduino IDE (HW stays the same) in menu for https://github.com/stm32duino/Arduino_Core_STM32/blob/5c87586d8cbefe314ccd74297e919c045335619c/variants/STM32F1xx/F103C8T_F103CB(T-U)/variant_generic.h  Serial monitor works, and USART pinout is the same....

fpistm, 2023-03-20:
No. For this board the default Serial instance is set to use those pins of Usart2.
If you read the wiki or search in issue or forum you would find the answer: https://github.com/stm32duino/Arduino_Core_STM32/wiki/API#hardwareserial

zlomennypez, 2023-03-20:
> 

> ~No. For this board the default Serial instance is set to use those pins of Usart2. If you read the wiki or search in issue or forum you would find the answer: ~ https://github.com/stm32duino/Arduino_Core_STM32/wiki/API#hardwareserial
> 
> OK. Read to quickly. If same usart works with other variant, it is probably a system click config issue. You should try to change it to test. See wiki.

I think, the Serial ports should be  same as Bluepill STMF103CB. In the variant_generic.h are the same, if I look at correct file, but It´s not working and I dont know why. It has something with programming. If I use putty and read the communication, there is nothing, even if Serial is correctly defined...

fpistm, 2023-03-20:
That's why I said the clock config has probably an issue.

fpistm, 2023-03-21:
Hi @zlomennypez 
I've tested and default `Serial `instance mapped on `Serial2 `(TX PA2) works as expected, I've also tested with USB CDC enabled and Serial mapped on SerialUSB works also.
So you probably have an issue with your hardware. 
Which board you have?
What is your sketch? Test with a basic one. I saw on your screenshot you have some FLASH constant so maybe you sketch is not compatible with STM32F103RE while is with STM32F103ZE.

zlomennypez, 2023-03-21:
> Hi @zlomennypez I've tested and default `Serial `instance mapped on `Serial2 `(TX PA2) works as expected, I've also tested with USB CDC enabled and `Serial `mapped on `SerialUSB `works also. So you [...]

Thank you, I will try to find a problem. I have custom board.

fpistm, 2023-03-22:
Well, except if it is a clone, I don't understand why it works with one variant and not with other. Sound very strange.
If you add a blink in your sketch, is the led blink? if not probably the program stuck in an error handler, HSI init too long? or something else.. hard to tell.

closed 2023-03-27 by fpistm

zlomennypez, 2023-03-27:
> Komu: stm32duino/Arduino_Core_STM32 ***@***.***>
> com>
> Stm32F103R (Issue #1980)
> "
>
> Closed #1980(https://github.com/stm32duino/Arduino_Core_STM32/issues/1980)
> as completed.
>
> (https://github.com/stm32duino/Arduino_Core_STM32/issues/1980#event-8850529492)
> "

Hello,

its blinking normally, the only thing, which is not working, is Serial 
Monitor. And when I choose STM32F1C, its blinking and monitor also works.

fpistm, 2023-03-27:
Which variant allow you to have serial?

zlomennypez, 2023-03-27:
Bluepill F103CB (or C8 with 128k) . My code is smaller than 128k, so it shows that it is filled from 80%, but works normally. Chip should be original STM32F103RET6 from JLCPCB.com

zlomennypez, 2023-03-27:
Looks like this 
[image]

zlomennypez, 2023-03-28:
STM32F103Z also works, but I have to test it.

zlomennypez, 2023-03-29:
VCCGND F103ZET6 looks like working without any problems

fpistm, 2023-03-30:
Pay attention VCC GND uses PA9/PA10 while generic uses PA2/PA3 for default Serial instance.