stm32duino/Arduino_Core_STM32 #1982: STM32H750XB

wystewart, 2023-03-25:
Please add support for the STM32H750XB

perhaps with some direction/help I could modify the 750VB files to work

wystewart, 2023-03-25:
oh looks like it might already be here: https://github.com/stm32duino/Arduino_Core_STM32/tree/main/variants/STM32H7xx/H742X(G-I)H_H743X(G-I)H_H745X(G-I)H_H747X(G-I)H_H750XBH_H753XIH_H755XIH_H757XIH

wystewart, 2023-03-25:
files seem to be there in installed 2.4.0 package, but not sure how to select from Arduino IDE

wystewart, 2023-03-25:
seems this needs to be added to boards.txt.  Adding manually, will report back when I have tried to compile with it.

fpistm, 2023-03-25:
Hi @wystewart 
yes all mcu have their directory including all generic files, they are generated automatically. But it is not enough to get it available in Arduino menu and build:
https://github.com/stm32duino/Arduino_Core_STM32/wiki/Add-a-new-variant-%28board%29#define-a-new-generic-variant

So you have to add ldscript.ld, a generic clock config and finally declare it in the boards.txt.

I added it to the variant request list #722

Do not hesitate to provide a PR if you made it.

PR #2158 (2023-10-31): variant(H7): add generic H747XIH and STM32H747I-DISCO
Fixes #1106
Fixes #1982
closed 2023-03-25 by fpistm