stm32duino/Arduino_Core_STM32 #1983: Wire stuck with MPU6050

EeeLo, 2023-03-27:
when I use this lib by arduino to build MPU6050 read data, the Wire lib stuck as any other people's problem,in Arduino's Wire lib, using Wire.setWireTimeout (time, researt) can researt tht I2C comm, how can I do it in this Wire lib? Wire.setTimeout doesn't work with me.

fpistm, 2023-03-27:
`setTimeout `is not implemented in this core.
Some Wire timeout api have been added later to Wire API as described by the portability notes of the API:
https://reference.arduino.cc/reference/en/language/functions/communication/wire/setwiretimeout/
We do not defined `WIRE_HAS_TIMEOUT`.

You can try the `recoverBus()` API anyway maybe you got another issue. Do you wired Pull-Up resistors on each I2C lines?

EeeLo, 2023-03-27:
Thanks for your replay, I know this question is about hardware error, it can run well when I run it with Arduino board whit auto restart,so I want to do that with nucleo stm32 board,but I found there have no member named "setWireTimeout" function,how can I open it with macro? And how can I using recoverBus() when I2C already in while() stuck?

fpistm, 2023-03-27:
Wire have a Timeout internally, so you should be able to call `Wire.recoverBus() ` when got 2C_* errors.
Don't know your sketch nor if you used MPU6050  library so hard  to help. 
Anyway, seems not very efficient to have to always restart I2C with Arduino.

EeeLo, 2023-03-27:
The question is like this : https://github.com/jrowberg/i2cdevlib/issues/543
I'm using i2cdevlib to decode MPU6050's DMP data, but always stuck in the dmpGetCurrentFIFOPacket(fifoBuffer), a lot of questions like me befor and use Wire.setWireTimeout(3000, true) can fix it,I want to build project on STM32 board and the question is there have no Wire.setWireTimeout function can use, I don't know what can I do next.

EeeLo, 2023-03-27:
`#include "MPU6050_6Axis_MotionApps20.h"
//#include "MPU6050.h" // not necessary if using MotionApps include file

// Arduino Wire library is required if I2Cdev I2CDEV_ARDUINO_WIRE implementation
// is used in I2Cdev.h

#if I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE
    #include "Wire.h"
#endif

#include <Wire.h>

// class default I2C address is 0x68
// specific I2C addresses may be passed as a parameter here
// AD0 low = 0x68 (default for SparkFun breakout and InvenSense evaluation board)
// AD0 high = 0x69

MPU6050 mpu;
//MPU6050 mpu(0x69); // <-- use for AD0 high

/* =========================================================================
   NOTE: In addition to connection 3.3v, GND, SDA, and SCL, this sketch
   depends on the MPU-6050's INT pin being connected to the Arduino's
   external interrupt #0 pin. On the Arduino Uno and Mega 2560, this is
   digital I/O pin 2.
 * ========================================================================= */

/* =========================================================================
   NOTE: Arduino v1.0.1 with the Leonardo board generates a compile error
   when using Serial1.write(buf, len). The Teapot output uses this method.
   The solution requires a modification to the Arduino USBAPI.h file, which
   is fortunately simple, but annoying. This will be fixed in the next IDE
   release. For more info, see these links:

   http://arduino.cc/forum/index.php/topic,109987.0.html
   http://code.google.com/p/arduino/issues/detail?id=958
 * ========================================================================= */

// uncomment "OUTPUT_READABLE_QUATERNION" if you want to see the actual
// quaternion components in a [w, x, y, z] format (not best for parsing
// on a remote host such as Processing or something though)
//#define OUTPUT_READABLE_QUATERNION

// uncomment "OUTPUT_READABLE_EULER" if you want to see Euler angles
// (in degrees) calculated from the quaternions coming from the FIFO.
// Note that Euler angles suffer from gimbal lock (for more info, see
// http://en.wikipedia.org/wiki/Gimbal_lock)
//#define OUTPUT_READABLE_EULER

// uncomment "OUTPUT_READABLE_YAWPITCHROLL" if you want to see the yaw/
// pitch/roll angles (in degrees) calculated from the quaternions coming
// from the FIFO. Note this also requires gravity vector calculations.
// Also note that yaw/pitch/roll angles suffer from gimbal lock (for
// more info, see: http://en.wikipedia.org/wiki/Gimbal_lock)
#define OUTPUT_READABLE_YAWPITCHROLL

// uncomment "OUTPUT_READABLE_REALACCEL" if you want to see acceleration
// components with gravity removed. This acceleration reference frame is
// not compensated for orientation, so +X is always +X according to the
// sensor, just without the effects of gravity. If you want acceleration
// compensated for orientation, us OUTPUT_READABLE_WORLDACCEL instead.
//#define OUTPUT_READABLE_REALACCEL

// uncomment "OUTPUT_READABLE_WORLDACCEL" if you want to see acceleration
// components with gravity removed and adjusted for the world frame of
// reference (yaw is relative to initial orientation, since no magnetometer
// is present in this case). Could be quite handy in some cases.
//#define OUTPUT_READABLE_WORLDACCEL

// uncomment "OUTPUT_TEAPOT" if you want output that matches the
[... 205 lines cut ...]
            mpu.dmpGetLinearAccelInWorld(&aaWorld, &aaReal, &q);
            Serial1.print("aworld\t");
            Serial1.print(aaWorld.x);
            Serial1.print("\t");
            Serial1.print(aaWorld.y);
            Serial1.print("\t");
            Serial1.println(aaWorld.z);
        #endif
    
        #ifdef OUTPUT_TEAPOT
            // display quaternion values in InvenSense Teapot demo format:
            teapotPacket[2] = fifoBuffer[0];
            teapotPacket[3] = fifoBuffer[1];
            teapotPacket[4] = fifoBuffer[4];
            teapotPacket[5] = fifoBuffer[5];
            teapotPacket[6] = fifoBuffer[8];
            teapotPacket[7] = fifoBuffer[9];
            teapotPacket[8] = fifoBuffer[12];
            teapotPacket[9] = fifoBuffer[13];
            Serial1.write(teapotPacket, 14);
            teapotPacket[11]++; // packetCount, loops at 0xFF on purpose
        #endif

    }
}`

EeeLo, 2023-03-27:
There have no PU resistor on I2C lines.
That's very useful to add Wire Timeout API, I already try use I2C with 100Khz but no surprise.
Any good way to do that? like use another I2C library with timeout reset API?

EeeLo, 2023-03-27:
I add 5k resistor each way but question again

fpistm, 2023-03-27:
After reading the link you provide, I don't think it is the same problem or not exactly the same. Our Wire implementation does not have blocking while loop as all have a timeout management.
Even 2 timeout managements: one from the HAL and one from the Wire based on[ `I2C_TIMEOUT_TICK`](https://github.com/stm32duino/Arduino_Core_STM32/blob/5a3f412d385c386eb421ede38c4223dbab5d83df/libraries/Wire/src/utility/twi.c#L49)
So our lib should not block and library which use it should manage the recovery itself based on error code returned.

fpistm, 2023-03-27:
Finally, as I thought and described in the `setTimeout()` Arduino documentation:
`
Note: these timeouts are almost always an indication of an underlying problem, such as misbehaving devices, noise, insufficient shielding, or other electrical problems. These timeouts will prevent your sketch from locking up, but not solve these problems. In such situations there will often (also) be data corruption which doesn’t result in a timeout or other error and remains undetected. So when a timeout happens, it is likely that some data previously read or written is also corrupted. Additional measures might be needed to more reliably detect such issues (e.g. checksums or reading back written values) and recover from them (e.g. full system reset). This timeout and such additional measures should be seen as a last line of defence, when possible the underlying cause should be fixed instead.`

So if I resume, our Wire already have a timeout management, the only difference is that reset is not done automatically and I think it should not be done automatically as it will hide another issue.
It seems links to third party library which support MPU6050.

EeeLo, 2023-03-27:
Thanks very much for your support,I see that,that project using another I2C lib not this lib.

EeeLo, 2023-03-29:
Hello @fpistm , sorry for that, How can I use it to auto reset I2C when error from bus? I found the function endTransmission with false can do it in Wire.cpp, But I edit it to false but not working

uint8_t TwoWire::endTransmission(void)
{
  return endTransmission((uint8_t)**true**);
}

fpistm, 2023-03-29:
No the false is argument to set to not send stop byte and allowing repeated start. No link with a reset of the bus.

EeeLo, 2023-03-29:
Alright, thanks