stm32duino/Arduino_Core_STM32 #1992: STM32 Core 2.5 Installation Issue

HelgeSeidel, 2023-04-03:
Got the following error message during installation of STM32 Core 2.5 using Arduino Board Manager:

_Failed to install platform: STMicroelectronics:stm32.
Error: 13 INTERNAL: Cannot install tool STMicroelectronics:xpack-openocd@0.12.0-1: testing local archive integrity: testing archive checksum: archive hash differs from hash in index_

 - OS: MAC OS 13.3
 - Arduino IDE version: 2.0.4
 - STM32 core version: 2.5

fpistm, 2023-04-03:
Hi @HelgeSeidel 
thanks for the issue.
It should be fixed now: https://github.com/stm32duino/BoardManagerFiles/commit/e0fe506505689fb3879fa6509fe62b2ac7cf95c4

fpistm, 2023-04-03:
Json file probably not updated. Ide2 has some issue around that. You probably  have to delete manually some files.

closed 2023-04-03 by HelgeSeidel