stm32duino/Arduino_Core_STM32 #1998: STM32 JSON URL for Arduino doesn't work

rikilshah, 2023-04-10:
STM32 Arduino core's JSON url doesn't work when added in Arduino IDE's preferences. I have copied url from official guide which is as below: 

[https://github.com/stm32duino/BoardManagerFiles/raw/main/package_stmicroelectronics_index.json](https://github.com/stm32duino/BoardManagerFiles/raw/main/package_stmicroelectronics_index.json)

When I open this directly in Browser, it is not returning a JSON file, which is normal behaviour. If I go to Github of Boardmanager, that json is there but can't open "Raw" link from there too. 

[https://github.com/stm32duino/BoardManagerFiles/blob/main/package_stmicroelectronics_index.json](https://github.com/stm32duino/BoardManagerFiles/blob/main/package_stmicroelectronics_index.json)

Because of that, ARduino is not able to Install/Update STM32duino core. I reported similar issue for ESP32 and it got resolved by changing URL.

fpistm, 2023-04-10:
Raw link is correct. Second is to view it under GH page. 
Don't know why it does not work for you, maybe proxy issue .

closed 2023-04-10 by fpistm

rikilshah, 2023-04-10:
I found the issue to be specific to ISP, JioFiber from India. Similar issue of not able to open raw links was reported on Github for other repos. based on that, I used Google's Public DNS and issue resolved immediately.