stm32duino/Arduino_Core_STM32 #2008: after update to 2.5 : none-eabi-g++ Segmentation fault cc1plus

bobemoe, 2023-04-15:
Updated from STMicroelectronics:stm32 **2.4.0** to **2.5.0** and now I can't compile my project:
```
$ arduino-cli compile -e -b STMicroelectronics:stm32:GenF4:pnum=BLACKPILL_F411CE,rtlib=nanofps,xserial=none,usb=CDCgen -v

...

/root/.arduino15/packages/STMicroelectronics/tools/xpack-arm-none-eabi-gcc/12.2.1-1.2/bin/arm-none-eabi-g++ -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -DUSE_FULL_LL_DRIVER -mthumb @/tmp/arduino/sketches/8FD6856F4DEF37775A0655397FD10F3B/sketch/build.opt -c -Os -DNDEBUG -w -std=gnu++17 -ffunction-sections -fdata-sections -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -fno-use-cxa-atexit -I/drone/src/firmware_stm -I/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/cores/arduino/avr -I/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/cores/arduino/stm32 -I/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/cores/arduino/stm32/LL -I/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/cores/arduino/stm32/usb -I/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/cores/arduino/stm32/OpenAMP -I/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/cores/arduino/stm32/usb/hid -I/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/cores/arduino/stm32/usb/cdc -I/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/system/Drivers/STM32F4xx_HAL_Driver/Inc -I/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/system/Drivers/STM32F4xx_HAL_Driver/Src -I/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/system/STM32F4xx -I/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/system/Middlewares/ST/STM32_USB_Device_Library/Core/Src -I/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/system/Middlewares/OpenAMP -I/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/system/Middlewares/OpenAMP/open-amp/lib/include -I/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/system/Middlewares/OpenAMP/libmetal/lib/include -I/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/system/Middlewares/OpenAMP/virtual_driver -w -x c++ -E -CC -DSTM32F4xx -DARDUINO=10607 -DARDUINO_BLACKPILL_F411CE -DARDUINO_ARCH_STM32 "-DBOARD_NAME=\"BLACKPILL_F411CE\"" "-DVARIANT_H=\"variant_BLACKPILL_F411CE.h\"" -DSTM32F411xE -DUSBCON -DUSBD_VID=0 -DUSBD_PID=0 -DHAL_PCD_MODULE_ENABLED -DUSBD_USE_CDC -DHAL_UART_MODULE_ENABLED -DHWSERIAL_NONE -I/root/.arduino15/packages/STMicroelectronics/tools/CMSIS/5.7.0/CMSIS/Core/Include/ -I/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/system/Drivers/CMSIS/Device/ST/STM32F4xx/Include/ -I/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/system/Drivers/CMSIS/Device/ST/STM32F4xx/Source/Templates/gcc/ -I/root/.arduino15/packages/STMicroelectronics/tools/CMSIS/5.7.0/CMSIS/DSP/Include -I/root/.arduino15/packages/STMicroelectronics/tools/CMSIS/5.7.0/CMSIS/DSP/PrivateInclude -I/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/cores/arduino -I/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/variants/STM32F4xx/F411C(C-E)(U-Y) /tmp/arduino/sketches/8FD6856F4DEF37775A0655397FD10F3B/sketch/SrcWrapper.cpp -o /dev/null
Error while detecting libraries included by /tmp/arduino/sketches/8FD6856F4DEF37775A0655397FD10F3B/sketch/SrcWrapper.cpp
Generating function prototypes...
/root/.arduino15/packages/STMicroelectronics/tools/xpack-arm-none-eabi-gcc/12.2.1-1.2/bin/arm-none-eabi-g++ -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -DUSE_FULL_LL_DRIVER -mthumb @/tmp/arduino/sketches/8FD6856F4DEF37775A0655397FD10F3B/sketch/build.opt -c -Os -DNDEBUG -w -std=gnu++17 -ffunction-sections -fdata-sections -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -fno-use-cxa-atexit -I/drone/src/firmware_stm -I/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/cores/arduino/avr -I/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/cores/arduino/stm32 -I/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/cores/arduino/stm32/LL -I/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/cores/arduino/stm32/usb -I/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/cores/arduino/stm32/OpenAMP -I/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/cores/arduino/stm32/usb/hid -I/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/cores/arduino/stm32/usb/cdc -I/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/system/Drivers/STM32F4xx_HAL_Driver/Inc -I/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/system/Drivers/STM32F4xx_HAL_Driver/Src -I/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/system/STM32F4xx -I/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/system/Middlewares/ST/STM32_USB_Device_Library/Core/Src -I/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/system/Middlewares/OpenAMP -I/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/system/Middlewares/OpenAMP/open-amp/lib/include -I/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/system/Middlewares/OpenAMP/libmetal/lib/include -I/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/system/Middlewares/OpenAMP/virtual_driver -w -x c++ -E -CC -DSTM32F4xx -DARDUINO=10607 -DARDUINO_BLACKPILL_F411CE -DARDUINO_ARCH_STM32 "-DBOARD_NAME=\"BLACKPILL_F411CE\"" "-DVARIANT_H=\"variant_BLACKPILL_F411CE.h\"" -DSTM32F411xE -DUSBCON -DUSBD_VID=0 -DUSBD_PID=0 -DHAL_PCD_MODULE_ENABLED -DUSBD_USE_CDC -DHAL_UART_MODULE_ENABLED -DHWSERIAL_NONE -I/root/.arduino15/packages/STMicroelectronics/tools/CMSIS/5.7.0/CMSIS/Core/Include/ -I/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/system/Drivers/CMSIS/Device/ST/STM32F4xx/Include/ -I/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/system/Drivers/CMSIS/Device/ST/STM32F4xx/Source/Templates/gcc/ -I/root/.arduino15/packages/STMicroelectronics/tools/CMSIS/5.7.0/CMSIS/DSP/Include -I/root/.arduino15/packages/STMicroelectronics/tools/CMSIS/5.7.0/CMSIS/DSP/PrivateInclude -I/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/cores/arduino -I/root/.arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/variants/STM32F4xx/F411C(C-E)(U-Y) /tmp/arduino/sketches/8FD6856F4DEF37775A0655397FD10F3B/sketch/firmware_stm.ino.cpp -o /tmp/arduino/sketches/8FD6856F4DEF37775A0655397FD10F3B/preproc/ctags_target_for_gcc_minus_e.cpp
arm-none-eabi-g++: internal compiler error: Segmentation fault signal terminated program cc1plus
Please submit a full bug report, with preprocessed source (by using -freport-bug).
See <https://gcc.gnu.org/bugs/> for instructions.
```

**To Reproduce**
This is our docker CI build container, created like:
```
FROM scratch
ADD ./chimaera-minbase /
CMD ["/bin/bash"]
RUN apt update && apt install -y curl python3 python3-pip p7zip-full
RUN curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh | sh
RUN pip3 install pyserial
RUN arduino-cli config init
RUN arduino-cli config add board_manager.additional_urls https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
RUN arduino-cli config add board_manager.additional_urls https://github.com/stm32duino/BoardManagerFiles/raw/main/package_stmicroelectronics_index.json
RUN arduino-cli core update-index
RUN arduino-cli core install esp32:esp32
RUN arduino-cli core install STMicroelectronics:stm32
RUN ln -s /usr/bin/python3 /usr/bin/python
```
`chimaera-minbase` is Devuan (Debian) linux debootstrap created like:
```
debootstrap --variant=minbase --include=iproute2 chimaera ./chimaera-minbase http://deb.devuan.org/merged
```

Steps to reproduce the behavior:
1. Build the container shown above
2. create a simple hello world/blink sketch
3. run `arduino-cli compile -e -b STMicroelectronics:stm32:GenF4:pnum=BLACKPILL_F411CE,rtlib=nanofps,xserial=none,usb=CDCgen -v`

**Expected behavior**
Compiles as before the update.

**Workaround**
Downgrade core to 2.4 by pinning version in Dockerfile:
```
RUN arduino-cli core install STMicroelectronics:stm32@2.4
```

fpistm, 2023-04-15:
Unfortunately, it seems the new arm none eabi toolchain has an issue on your environment. 
Is it reproducible each time? And with other sketch? Or other board?

If yes you should file an issue to xpack arm none eabi gcc: https://github.com/xpack-dev-tools/arm-none-eabi-gcc-xpack

bobemoe, 2023-04-16:
Yes fully reproducible even with a simple blink sketch or a complex project. An ESP32 project compiles just fine on the container.
I will report to xpack-dev-tools. Thanks.

ilg-ul, 2023-04-16:
The problem seems to be in the processor, when the file is turned from `.ino.cpp` into plain `.cpp`. (sorry for my ignorance, I don't know what `.ino.cpp` files are).

Add a `-v` when invoking the compiler, and you'll get a more detailed output with the internal steps.

bobemoe, 2023-04-18:
Solved by Docker host upgrade.
Kernel from: 4.19.0-14-amd64 SMP Debian 4.19.171-2 (2021-01-30) x86_64 GNU/Linux
To: 5.10.0-21-amd64 SMP Debian 5.10.162-1 (2023-01-21) x86_64 GNU/Linux
Full discussion in https://github.com/xpack-dev-tools/arm-none-eabi-gcc-xpack/discussions/28

closed 2023-04-18 by bobemoe