stm32duino/Arduino_Core_STM32 #2052: STM32 MCU based boards v2.6.0 causes issue with STM32SD interface on STEVAL-MKSBOX1V1

gemelko, 2023-07-06:
I upgraded to v2.6.0 today and the SD.begin() call in the STM32SD library stalls. 

There was a bug in STM32SD v1.3.0 which caused the STM32SD to stall, which I reported, resolved here: https://github.com/stm32duino/STM32SD/issues/60

However the board package v2.6.0 causes SD.begin() to stall under both STM32SD v1.3.0[patched] AND v1.2.3 so I had to downgrade to v2.5.0 to allow my project to continue to run. After downgrade to v2.5.0 both STM32SDv1.3.0[patched] and v1.2.3 work again.

Glenn

fpistm, 2023-07-06:
@gemelko 
As stated in the issue you have submitted to STM32SD, I told you there is another issue so please avoid to open several issues on the same topic.

closed 2023-07-06 by fpistm

gemelko, 2023-07-06:
It felt similar yet somehow also different, as this release of the board package impacted both versions of the STM32SD library. I was not aware that the same ppl are working on both. Thanks again: overall the support for STM32duino is amazing!

Glenn

PR #2053 (2023-07-10): Sdmmc signals
Fixes #2052

https://github.com/stm32duino/STM32SD/releases/tag/1.3.1 already allows to fix this, this PR implements a better granularity.