stm32duino/Arduino_Core_STM32 #2055: Serial.available() and Serial.read() not working with USB CDC virtual COM port

vibhoregit, 2023-07-08:
**Describe the bug**
Serial loopback with USB virtual COM port is not working as Serial.available() returns 0 always.

**To Reproduce**
`void setup() {
  // Initialize the USB serial port
  Serial.begin(9600);
  pinMode(PC13, OUTPUT);
  digitalWrite(PC13, HIGH);
  while (!Serial);
  Serial.println(F("USB Serial Initialised"));
}

void loop() {
  // Check if any characters are available to read
  if (Serial.available()) {
    char ch = Serial.read();  // Read one character from serial buffer
    Serial.print(ch);         // Print the character back to serial monitor
  }
}
`

Steps to reproduce the behavior:
1. Upload to blue pill board with either FTDI/Serial adapter or rogerclarke's [bootloader](https://github.com/rogerclarkmelbourne/STM32duino-bootloader/blob/master/binaries/generic_boot20_pc13.bin)
2. Open COM port of the blue pill board which appears on connecting the board via its USB port
3. Observe "USB Serial Initialised" to get printed but any character sent from serial monitor is not echoed back.

**Expected behavior**
"USB Serial Initialised" gets printed and whatever characters are entered in the serial monitor should echo back

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Arduino IDE version: 2.1.0
 - STM32 core version: 2.6.0
 - Tools menu settings if not the default: Available in screenshot
 - Upload method: Tried STM32duino (Serial) and Maple bootloader 2.0 

**Board: Bluepill**
 - Name: Generic STM32F1 Series
 - Hardware Revision: Latest Revision (with correct resistor (1.5k) on D+ pin)
 - Extra hardware used if any: No extra hardware

**Additional context**
I tried a bunch of things to narrow down the problem so that actual problem can be pin pointed

1. Uploaded CDC example code compiled on Keil microvision to test if windows drivers are correct: The board is able to transmit and receive data over USB and COM port driver seems to work fine.
2. Tested Serial library with Serial1 to test if hardware serial works: The Arduino sketch works with Serial1 as expected and board echos back the data sent over serial adapter to its UART1.
3. Tested the board with android device: CDC device is recognized and **Serial USB Terminal** app is able to connect to the COM port but as soon as a character is sent, the app hangs for a second and the COM port disconnects.
4. Printing Serial.available() returns 0 to Serial monitor no matter how many characters are sent.

fpistm, 2023-07-10:
Hi @vibhoregit 
I've tested with your sketch and my BluePill F103C8 and it works as expected. It fails only if it is large amount of data as stated in #1399.
Ensure you have a legacy mcu and a correct table.

closed 2023-07-10 by fpistm

bmo, 2024-04-21:
I am also able to reproduce this issue, even when the "SerialUSB" device is used.
Serial writes work, however serial reads do NOT work. No characters are ever available. I've tried both with USB being the generic serial, and with it NOT being generic serial. 

Desktop (please complete the following information):

OS: MacOs
Arduino IDE version: 2.3.2
STM32 core version: 2.7.1
Tools menu settings if not the default: Available in screenshot
Upload method: ST-Link clone on HW port

zougloub, 2024-06-14:
I purchased some Blue pills and recently noticed too late that they contain STM32F103 clones, and exhibit this issue with Serial.available() in the Arduino core. I noticed that a workaround using Serial.read() == -1 would  allow reads...

Hackin7, 2024-06-16:
I'm having a similar issue too, serial reads do not work

Yurisu, 2024-06-29:
Whatever character sending Serial.read() is reads -1, and Serial.available() does not work

Art-ut-Kia, 2024-10-19:
I felt accross the problem with a BluePill that reports 256kB Flash, High Density (obviously a fake STM32).
Did anyone find a work around?

zougloub, 2024-10-19:
I didn't want to spend too much time investigating this, but if someone has time to lose vs. simply purchasing genuine hardware, I'd suggest:

- Making a "hello F103 USB CDC world" project in STM32CubeIDE, for confirming wiether it doesn't work there (behavior NG on clone / OK on genuine), as STM32Cube BSP is used for Arduino, would be a good idea.

- I *think* there's something that works in some open source projects based on BSPs that aren't recent STM32Cube ones. The clones don't bother providing software or proper documentation for their chips, they're just ripping off ST's work (which should be sufficient for avoiding them... now there are also people that got them unknowingly) and I guess it's probable the hardware works with KEIL or other "mainstream" IDEs that are also commonplace in China.

Art-ut-Kia, 2024-10-19:
Meanwhile I changed from STMicroelectronics core to STM32duino core.
The serial over USB works well with the latter, bidirectional.

zougloub, 2024-10-19:
Oh interesting. Would you mind elaborating on « I changed from STMicroelectronics core to STM32duino core », maybe I'll take some time to check further... I don't like having a handful of unusable boards around.

Art-ut-Kia, 2024-10-19:
[image]

Art-ut-Kia, 2024-10-19:
I first used the "STM32 MCU based boards by STMicroelectronics" core with which SerialUSB.available() and SerialUSB.read() didn't work.
I then give a try to the "STM32F1xx/GDF1xx boards by stm32duino". Serial over USB is now named Serial, and it works!

Art-ut-Kia, 2024-10-19:
PS: in both cases, I didn't use a bootloader. I rather uploaded the programs with an ST-link V2

zougloub, 2024-10-20:
Just tested and indeed [the other stm32duino core](https://github.com/rogerclarkmelbourne/Arduino_STM32) is working on my fake F103 (thanks Amazon...), and with the corresponding bootloader. With this core, the USB enumeration doesn't work.

zougloub, 2024-10-20:
Made that STM32CubeIDE test (I wanted to document the steps but I used an IOC for CBT6, 72 MHz, and then it's pretty much https://deepbluembedded.com/stm32-usb-cdc-virtual-com-port-vcp-examples/) and it's also working fine.