stm32duino/Arduino_Core_STM32 #2056: stm32g0xx_hal_flash.h does not have `FLASH_FLAG_ALL_ERRORS`

amanenk, 2023-07-08:
**Describe the bug**
I am trying to create a program that uses flash memory of  STM32G030F6 as storage for configs. I have found a library that might do it but i cant build a code with it. the link to the library https://github.com/khoih-prog/FlashStorage_STM32. When i try to build the library the compiler gives the next error ` error: 'FLASH_FLAG_ALL_ERRORS' was not declared in this scope; did you mean 'FLASH_FLAG_REG_POS'?`. 

**To Reproduce**
to reproduce you need to create an empty project with STM32G030F6 , import #khoih-prog/FlashStorage_STM32 library, and try to build it.

fpistm, 2023-07-08:
Hi @amanenk ,
This issue is not linked to the core. This flag is defined by the HAL for some series and used by third party library.
As the core uses HAL without no change, this issue is not valid. Third party librariy have to update its code or submit this to official HAL tepo

closed 2023-07-08 by fpistm

fpistm, 2023-07-10:
Just to go further, the `FLASH_FLAG_ALL_ERRORS` was removed from the STM32CubeG0 v1.4.1, see #1253 .
As you can see I got the same issue and resolve this like this:
https://github.com/stm32duino/Arduino_Core_STM32/pull/1253/commits/b3f5a2f245f08f1e980ce43a961e7ad688aa33cb

So you can probably define it like I do.