stm32duino/Arduino_Core_STM32 #2070: Use CDC for USB HS

andreych2222, 2023-07-14:
The USB header files contain define USE_USB_HS. But it will not work to initialize CDC for HS, since Id = 0 is written in the CDC_Init function.

void CDC_init(void)
{
  if (!CDC_initialized) {
    /* Init Device Library */
    if (USBD_Init(&hUSBD_Device_CDC, &USBD_Desc, 0) == USBD_OK) ...

fpistm, 2023-07-14:
Hi @andreych2222 could you be more precise. 
Which board you used? Arduino IDE version ? Core version ? All relevant information to help you..

andreych2222, 2023-07-17:
I used PlatformIO 6.1.9 with VSCode, framework-arduinoststm32 @ 4.20200.220530 (2.2.0). My board based on stm32f407zgt6. I selected board = black_f407zg. My board has two usb - FS and HS. 

I defined 
build_flags =   -D USBCON
                        -D USBD_USE_CDC

and 
#include <USBSerial.h>
....
SerialUSB.begin();
....

Everything is fine, USB FS works. But I want to have CDC on USB HS.

In file ...\framework-arduinoststm32\cores\arduino\stm32\usb\cdc\usbd_cdc_if.c I see

#ifdef **USE_USB_HS**
  #define CDC_MAX_PACKET_SIZE USB_OTG_HS_MAX_PACKET_SIZE
#elif defined(USB_OTG_FS) || defined(USB_OTG_FS_MAX_PACKET_SIZE)

and in the function i see

void CDC_init(void)
{
  if (!CDC_initialized) {
    /* Init Device Library */
    if (**USBD_Init**(&hUSBD_Device_CDC, &USBD_Desc, **0**) == USBD_OK) {.....

As far as I understand, 1 should be passed to the function USBD_Init for USB HS.

Maybe I don't understand something, how can I use USB HS CDC?

fpistm, 2023-07-17:
Hi, thanks for the feedback.
`id` is not link to HS. 
I've tested with a disco F746 and it works as expected:
[image]

About your define, it seems not enough to define USBD as HS. In Arduino you have to select the USB speed:
[image]

which define: `-DUSE_USB_HS`
anyway, here we do not support PIO. And as stated for other issue here, PIO is not up to date with the last core version.

closed 2023-07-17 by fpistm