stm32duino/Arduino_Core_STM32 #2073: STM32F103 - 'TIM4' was not declared in this scope

Contomo, 2023-07-16:
Hello, 
none of the F103 boards seem to compile with Timer4
in the attatched code, PB6 maps to TIM4 channel 1, the code as posted compiles, but crashes the microcontroller, changing the 

HardwareTimer *TimerSteppers = new HardwareTimer(Instance);
to:
HardwareTimer *TimerSteppers = new HardwareTimer(TIM4);
it doesnt compile anymore as it cant find TIM4.

The code posted was tested on an STM32F411, compiled fine, uploaded fine, and ran well (until i accidentally killed the chip)

i tried updating to the latest versioon, using arduino 1.8.13
(Timer 5 and 6 also are unavalible)
am i doing something fundamentally wrong? or is there actually something missing?

```
#include <TMC2209.h>

//-----------------------Stepper driver definitions-----------------------

#define StepPin0 PB6
#define StepPin1 PB7
#define StepPin2 PB8
#define StepPin3 PB9

#define DirPin0 PB12
#define DirPin1 PB13
#define DirPin2 PB14
#define DirPin3 PB15

#define AccX 10000
#define AccY 10000
#define AccZ 10000
#define AccA 10000

#define MaxX 100
#define MaxY 200
#define MaxZ 200
#define MaxA 200

#define NUM_STEPPERS 4

#define MicroStepsX 6
#define MicroStepsY 6
#define MicroStepsZ 6
#define MicroStepsA 6

TMC2209 stepper_driver_0;
const TMC2209::SerialAddress SERIAL_ADDRESS_0 = TMC2209::SERIAL_ADDRESS_0;
TMC2209 stepper_driver_1;
const TMC2209::SerialAddress SERIAL_ADDRESS_1 = TMC2209::SERIAL_ADDRESS_1;
TMC2209 stepper_driver_2;
const TMC2209::SerialAddress SERIAL_ADDRESS_2 = TMC2209::SERIAL_ADDRESS_2;
TMC2209 stepper_driver_3;
const TMC2209::SerialAddress SERIAL_ADDRESS_3 = TMC2209::SERIAL_ADDRESS_3;

const uint8_t REPLY_DELAY = 50;

struct stepperInfo {
  // externally defined parameters
  float acceleration;
  volatile unsigned long minStepInterval; // ie. max speed, smaller is faster
  void (*dirFunc)(int);
  void (*stepFunc)();

  // derived parameters
  unsigned int c0;                // step interval for first step, determines acceleration
  long stepPosition;              // current position of stepper (total of all movements taken so far)

  // per movement variables (only changed once per movement)
[... 420 lines cut ...]
    stepper_driver_2.setRunCurrent(RUN_CURRENT_PERCENT_2);
    stepper_driver_2.enableCoolStep();
    stepper_driver_2.setMicrostepsPerStepPowerOfTwo(microsteps_per_step_exponent);
    stepper_driver_2.enableAutomaticCurrentScaling();
    stepper_driver_2.enableAutomaticGradientAdaptation();
    stepper_driver_2.setMicrostepsPerStepPowerOfTwo(MicroStepsZ);
    stepper_driver_2.enable();
    delay(10);
  }

  if (Driver == 3 || Driver == -1) {

    stepper_driver_3.setup(Serial2, SERIAL_BAUD_RATE, SERIAL_ADDRESS_3);
    stepper_driver_3.setReplyDelay(REPLY_DELAY);
    stepper_driver_3.setRunCurrent(RUN_CURRENT_PERCENT_3);
    stepper_driver_3.enableCoolStep();
    stepper_driver_3.setMicrostepsPerStepPowerOfTwo(microsteps_per_step_exponent);
    stepper_driver_3.enableAutomaticGradientAdaptation();
    stepper_driver_3.enableAutomaticCurrentScaling();
    stepper_driver_3.setMicrostepsPerStepPowerOfTwo(MicroStepsA);
    stepper_driver_3.enable();
    delay(10);
  }
}
```

fpistm, 2023-07-17:
Hi @Contomo 
Don't know how you met this but `TIM4 `is well defined for mcu with it.

https://github.com/stm32duino/Arduino_Core_STM32/blob/main/system/Drivers/CMSIS/Device/ST/STM32F1xx/Include/stm32f103xb.h#L650
https://github.com/stm32duino/Arduino_Core_STM32/blob/8ff274affbbce42f1d452e74365b66ba03e66c6c/system/Drivers/CMSIS/Device/ST/STM32F1xx/Include/stm32f103xe.h#L836
https://github.com/stm32duino/Arduino_Core_STM32/blob/8ff274affbbce42f1d452e74365b66ba03e66c6c/system/Drivers/CMSIS/Device/ST/STM32F1xx/Include/stm32f103xg.h#L849

Only this one has no TIM4:
https://github.com/stm32duino/Arduino_Core_STM32/blob/main/system/Drivers/CMSIS/Device/ST/STM32F1xx/Include/stm32f103x6.h

Maybe you select the wrong target.

closed 2023-07-17 by fpistm

Contomo, 2023-07-17:
[image]
found in C:\Users\UserName\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.6.0\variants\STM32F1xx\F103C4T_F103C6(T-U)

why is there no pins mapped to timer4? like PB6-PB9 dont exist

Contomo, 2023-07-17:
ok yeah definetly me being sleep deprived, sorry for that, apparently my microcontroller actually has no TIM4....

fpistm, 2023-07-17:
Those files are automatically generated from the https://github.com/STMicroelectronics/STM32_open_pin_data files.
So if not present here, they are not referenced for this mcu that's all.
If you look at the datasheet: https://www.st.com/en/microcontrollers-microprocessors/stm32f103c6.html
there is no TIM4.

fpistm, 2023-07-17:
> [image] found in C:\Users\UserName\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.6.0\variants\STM32F1xx\F103C4T_F103C6(T-U)
> 
> why is there no pins mapped to timer4? like PB6-PB9 dont exist