stm32duino/Arduino_Core_STM32 #2077: Example does not compile, core 2.6.0

brightproject, 2023-07-20:
Example from [here](https://github.com/stm32duino/Arduino_Core_STM32/wiki/API#hardwareserial) doesn't compile in `Arduino IDE 1.8.19`

```
//                      RX    TX
HardwareSerial Serial1(PA10, PA9);

void setup() {
  Serial1.begin(115200); 
}

void loop() {
  Serial1.println("Hello World!");
  delay(1000);
}
```

> c:/users/admin/appdata/local/arduino15/packages/stmicroelectronics/tools/xpack-arm-none-eabi-gcc/12.2.1-1.2/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\admin\AppData\Local\Temp\arduino_cache_150181\core\core_a9c0fc81ed87578aed206a502d28436a.a(HardwareSerial.cpp.o):(.bss.Serial1+0x0): multiple definition of `Serial1'; sketch\MPU9250_DMP_Quaternion.ino.cpp.o:(.bss.Serial1+0x0): first defined here
> collect2.exe: error: ld returned 1 exit status
> exit status 1
> 
Related [link](https://github.com/stm32duino/Arduino_Core_STM32/discussions/2076).

brightproject, 2023-07-20:
> it seems you don't understand how `Serial `instance is defined.

Yes, you are right, until recently I did not quite understand UART.
It turns out that I cannot use the `I2C` bus on `PB6` and P`B7` at the same time, I cannot use `UART1`?

fpistm, 2023-07-20:
> It turns out that I cannot use the `I2C` bus on `PB6` and `PB7` at the same time(simultaneously), I cannot use `UART1`?

You can, but in all your example there is no I2C code using Wire library.
Default pins for Wire are PB6/PB7:
https://github.com/stm32duino/Arduino_Core_STM32/blob/96d8c937d8b801336aa5348cbb524887abc611aa/variants/STM32F4xx/F401CC(F-U-Y)_F401C(B-D-E)(U-Y)/variant_BLACKPILL_F401Cx.h#L111-L117

So simply use `Serial `instance which using PA9/PA10 and `Wire `instance which using PB6/PB7.
If you use one of the default Wire example like I2C scanner, you will see your I2C device address print on the Serial.

fpistm, 2023-07-20:
Honestly, I don't understand your issue.
What it exactly your code? I2C and Serial works properly on blackPill F4. I've one and have no issue.

fpistm, 2023-07-20:
Well, I've got an MPU-9250 an it works as expected with a blackPill F4 and the MPU9250_Basic.ino and it works as expected:

> 18:57:39.331 -> Accel: 0.80, -0.59, -0.15 g
> 18:57:39.331 -> Gyro: 34.63, -25.06, 99.09 dps
> 18:57:39.331 -> Mag: 55.51, -41.86, 7.35 uT
> 18:57:39.377 -> Time: 63826 ms
> 18:57:39.377 -> 
> 18:57:39.470 -> Accel: 0.73, -0.65, -0.16 g
> 18:57:39.470 -> Gyro: -26.89, 5.37, 17.01 dps
> 18:57:39.470 -> Mag: 57.91, -36.91, 6.75 uT
> 18:57:39.470 -> Time: 63926 ms
> 18:57:39.470 -> 
> 18:57:39.563 -> Accel: 0.97, -0.33, -0.12 g
> 18:57:39.563 -> Gyro: -15.24, 25.00, -174.21 dps
> 18:57:39.563 -> Mag: 57.01, -37.51, 6.90 uT
> 18:57:39.563 -> Time: 64026 ms
> 18:57:39.563 -> 
> 18:57:39.655 -> Accel: 0.93, -0.18, -0.13 g
> 18:57:39.655 -> Gyro: -24.51, -2.01, -49.70 dps
> 18:57:39.655 -> Mag: 45.01, -46.51, 6.90 uT
> 18:57:39.655 -> Time: 64126 ms
> 18:57:39.655 -> 
> 18:57:39.733 -> Accel: 0.75, -0.58, -0.26 g
> 18:57:39.733 -> Gyro: 15.00, -40.55, 163.35 dps
> 18:57:39.733 -> Mag: 45.31, -48.01, 6.90 uT
> 18:57:39.733 -> Time: 64226 ms
> 18:57:39.778 -> 
> 18:57:39.857 -> Accel: 0.76, -0.67, -0.23 g
> 

I've also tested with https://github.com/hideakitai/MPU9250 and it works too.

fpistm, 2023-07-20:
About:

> Wire.setClock(400000);

It have to be called after the `Wire.begin()` but as it is encapsulate in the Sparkfun library you have to edit it or kept 100kHz.

brightproject, 2023-07-20:
> Well, I've got an MPU-9250 an it works as expected with a blackPill F4 and the MPU9250_Basic.ino and it works as expected:

Thank you very much for taking the time to check.
I know for sure that my `MPU9250` is [fake](https://github.com/hideakitai/MPU9250/issues/93) and does not have a magnetometer.
But ... now I am conducting an experiment, I take `ESP32 + MPU9250` and `STM32F4 + MPU9250`.
I flash these two microcontrollers with the same code, of course, for `STM32` I choose the [core](https://github.com/stm32duino/Arduino_Core_STM32), and for `ESP32` I choose the [core](https://github.com/espressif/arduino-esp32).
These cores have different `wire.cpp` files.
But the point is not that, on each microcontroller, I compare with the help of a `logic analyzer` what happens on the `I2C` bus when the microcontroller communicates with the `MPU9250` board.
So, when I run the code on `STM32 + MPU9250`, packet transmission starts on the `I2C` bus.
The first packet has `3 bytes`, then after `99.6ms` another packet of `31 bytes`, and then after `2.9ms` the last packet of `30 bytes` and with that all activity on the bus stops.
In this case, of course, the `MPU9250` does not transmit any data to the bus, and nothing is output to the serial.
[image]
It is impossible to make screenshots in detail, because it is very small and you get a lot of information.
When I load the code into `ESP32 + MPU9250`, miracles happen.
Firstly, the data from the `MPU9250` board goes to the `serial` port.
But, what is going on on the` I2C` bus?
And what happens on the bus is approximately the same as on the `STM32F4` bus, only there are differences.
[image]
Packet transmission also starts with `3 bytes`, and is similar to the packet in `STM32F4`.
Then, after `99.3 ms`, there is also a packet of `31 bytes` and it is similar to the packet in `STM32F4`.
But the next packet is different ... it consists of `26 bytes`, i.e. `4 bytes` shorter than `STM32F4`.
Data from the signal analyzer of the third package from `STM32F4 + MPU9250`
[image]
Data from the signal analyzer of the third package from `ESP32 + MPU9250`
[image]
For `STM32F4`, communication on the `I2C` bus ends with `3 packets`.
For `ESP32` after three packets, activity on the A12C bus continues .
After the third packet, after `49.9 ms` there is a packet consisting of `16 bytes`, and then, also after `49.9 ms`, another packet consisting of `16 bytes`.
[image]
Well, then, after `50.1 ms`, the actual data transfer from the `MPU9250` begins and this data is output to the `ESP32 serial port`.
[image]
I don't understand what's wrong with my `STM32F4` or `MPU9250`?
The `I2C` bus works, but stops working.
@fpistm everything [works](https://github.com/stm32duino/Arduino_Core_STM32/issues/2077#issuecomment-1644272944) for you ...