stm32duino/Arduino_Core_STM32 #2078: STM32G070RBT6 can not run serial port 4 and the other three serial ports at the same time, otherwise the system will crash

ShiPC123, 2023-07-21:
**Please, before reporting any issue**
- Make sure you are using the latest Arduino_Core_STM32 version.
  https://github.com/stm32duino/Arduino_Core_STM32/releases/latest
- Make sure the issue is not already reported/fixed on GitHub or discussed on the [stm32duino forum](https://stm32duino.com)
- Submit a GitHub issue only for reporting a problem related to the [Arduino_Core_STM32](https://github.com/stm32duino/Arduino_Core_STM32).
- **Avoid** to submit a GitHub issue for project troubleshooting.

Any questions/feedback/suggestions should be discussed on the [stm32duino forum](https://stm32duino.com):
  * questions on the [STM32 Core](https://www.stm32duino.com/viewforum.php?f=35)
  * bugs/enhancements on the [STM core: Bugs and enhancements](https://www.stm32duino.com/viewforum.php?f=38)

&#x26A0; When reporting any issue, please try to provide all relevant information to help on its resolution.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Complete source code which can be used to reproduce the issue. Please try to be as generic as possible (no extra code, extra hardware,...)

Steps to reproduce the behavior:
1. Plug
2. Press USER_BTN
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. Windows]
 - Arduino IDE version: [e.g. 1.8.8]
 - STM32 core version: [e.g. 1.4.0]
 - Tools menu settings if not the default: [e.g. Newlib Standard, No Serial]
 - Upload method: [e.g. SWD]

**Board (please complete the following information):**
 - Name: [e.g. Nucleo F429ZI]
 - Hardware Revision: [e.g. Rev B]
 - Extra hardware used if any: [e.g. X-Nucleo IKS01A1]

**Additional context**
Add any other context about the problem here.

ShiPC123, 2023-07-21:
STM32G070RBT6   can not run serial port 4 and the other three serial ports at the same time, otherwise the system will crash
  #include<HardwareSerial.h>

HardwareSerial Serial3(PB9, PB8);
#define my485_en1  PB5

HardwareSerial Serial2(PD6, PD5);
#define my485_en2  PB3

HardwareSerial Serial1(PA10, PA9);

void setup() {

  Serial1.begin(115200);
  Serial4.setRx(PC11);
  Serial4.setTx(PC10);
  Serial4.begin(115200);

  pinMode(my485_en1, OUTPUT);
  pinMode(my485_en2, OUTPUT);

  Serial3.begin(115200);
  Serial2.begin(115200);

}

void loop() {
  Serial4.println("AA5000");
  Serial3.println("AA5000");
  Serial2.println("AA5000");
  Serial1.println("AA5000");
  delay(200);

}

ShiPC123, 2023-07-21:
If I comment out Serial4, the system will reply, and if I don't make additional statements about the pin of serial 4, there will be no jams
I feel like the problem is
Serial4.setRx(PC11);
Serial4.setTx(PC10);
But my actual project, need to use the above two pins, manual, these two pins are also serial port pins

ShiPC123, 2023-07-21:
我使用的软件版本是2.4的版本

ShiPC123, 2023-07-21:
If I only use serial ports 1 and 4 the system works fine
Once you add serial port 2 and serial port 3, the system can not run, only run serial port 1, serial port 2, serial port 3 system can work normally

brightproject, 2023-07-21:
> If I only use serial ports 1 and 4 the system works fine

Do you have pins `PA0` and `PA1` busy?
It seems that they are used for [USART4](https://github.com/stm32duino/Arduino_Core_STM32/blob/96d8c937d8b801336aa5348cbb524887abc611aa/variants/STM32G0xx/G070RBT/PeripheralPins.c#L152C1-L171C3) on your microcontroller.

If you want to assign other pins to `Serial4`, then it seems that this can be done by an alternative [method](https://github.com/stm32duino/Arduino_Core_STM32/wiki/API#hardwareserial).
Another solution is to add a [build_opt.h](https://github.com/stm32duino/Arduino_Core_STM32/wiki/Customize-build-options-using-build_opt.h) file alongside your main .ino file with: `-DENABLE_HWSERIALx`.
You can try this code
```
HardwareSerial MyPort1(USART1);
HardwareSerial MyPort2(USART2);
HardwareSerial MyPort3(USART3);
HardwareSerial MyPort4(USART4);

void setup() {
MyPort1.begin(115200);
MyPort2.begin(115200);;
MyPort3.begin(115200);
MyPort4.begin(115200);
}

void loop() {
MyPort1.println("AA5000");
MyPort2.println("AA5000");
MyPort3.println("AA5000");
MyPort4.println("AA5000");
delay(200);
}
```

ShiPC123, 2023-07-21:
How to add build_opt.h, what code needs to be written inside, can you give me an example? In addition, I read the manual that the two pins PC10 and PC11 can be used as serial port 3 or serial port 4. I have no way to modify PCB now, so there is no way to connect the serial port to PA0 and PA1

fpistm, 2023-07-21:
Hi @ShiPC123 
Your code is not correct.
If you look at the PeripheralPins.c
https://github.com/stm32duino/Arduino_Core_STM32/blob/96d8c937d8b801336aa5348cbb524887abc611aa/variants/STM32G0xx/G070RBT/PeripheralPins.c#L152-L193

To get Serial4 using USART4 you have to use the ALT pins else you use the USART3:
Your code should be:

```C++
#include<HardwareSerial.h>

HardwareSerial Serial3(PB9, PB8);
#define my485_en1  PB5

HardwareSerial Serial2(PD6, PD5);
#define my485_en2  PB3

HardwareSerial Serial1(PA10, PA9);

void setup() {

  Serial1.begin(115200);
  Serial4.setRx(PC11_ALT1);
  Serial4.setTx(PC10_ALT1);
  Serial4.begin(115200);

  pinMode(my485_en1, OUTPUT);
  pinMode(my485_en2, OUTPUT);

  Serial3.begin(115200);
  Serial2.begin(115200);

}

void loop() {
  Serial4.println("AA5000");
  Serial3.println("AA5000");
  Serial2.println("AA5000");
  Serial1.println("AA5000");
  delay(200);
}
```

ShiPC123, 2023-07-21:
Thank you very much that the problem has been solved

fpistm, 2023-07-21:
Welcome. I know it can be confusing but the only way to provide all possibilities for each pins.

closed 2023-07-21 by fpistm