stm32duino/Arduino_Core_STM32 #2106: compilation error: STM32F103CC

phalgunimondal, 2023-08-17:
**arduino-cli compile -b STMicroelectronics:stm32:GenF4:pnum=BLACKPILL_F401CC dmd_monochrome**

In file included from C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_RGB.h:24,
                 from C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_RGB.cpp:9:
C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_STM32a.h:235:40: error: **'voidFuncPtr' has not been declared**
  235 |         virtual void initialize_timers(voidFuncPtr handler);
      |                                        ^~~~~~~~~~~
C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_STM32a.h:264:9: error: 'PortType' does not name a type
  264 |         PortType clk_clrmask = 0;
      |         ^~~~~~~~
C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_STM32a.h:265:9: error: 'PortType' does not name a type
  265 |         PortType clkmask = 0;
      |         ^~~~~~~~
C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_STM32a.h:275:9: error: 'OE_PWM_Polarity' does not name a type
  275 |         OE_PWM_Polarity OE_polarity = OE_PWM_POSITIVE;
      |         ^~~~~~~~~~~~~~~
C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_RGB.cpp: In constructor 'DMD_RGB_BASE::DMD_RGB_BASE(byte, uint8_t*, byte, byte, uint8_t*, byte, byte, bool, uint8_t, uint8_t, byte, byte)':
C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_RGB.cpp:32:9: error: 'OE_polarity' was not declared in this scope
   32 |         OE_polarity = OE_PWM_NEGATIVE;
      |         ^~~~~~~~~~~
C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_RGB.cpp:32:23: error: 'OE_PWM_NEGATIVE' was not declared in this scope
   32 |         OE_polarity = OE_PWM_NEGATIVE;
      |                       ^~~~~~~~~~~~~~~
C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_RGB.cpp: In member function 'virtual void DMD_RGB_BASE::init(uint16_t)':
C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_RGB.cpp:167:27: error: invalid conversion from 'void (*)()' to 'int' [-fpermissive]
  167 |         initialize_timers(scan_running_dmd_R);
      |                           ^~~~~~~~~~~~~~~~~~
      |                           |
      |                           void (*)()
C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_STM32a.h:235:52: note:   initializing argument 1 of 'virtual void DMD::initialize_timers(int)'
  235 |         virtual void initialize_timers(voidFuncPtr handler);
      |                                        ~~~~~~~~~~~~^~~~~~~
In file included from C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_STM32a.cpp:27:
C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_STM32a.h:235:40: error: 'voidFuncPtr' has not been declared
  235 |         virtual void initialize_timers(voidFuncPtr handler);
      |                                        ^~~~~~~~~~~
C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_STM32a.h:264:9: error: 'PortType' does not name a type
  264 |         PortType clk_clrmask = 0;
      |         ^~~~~~~~
C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_STM32a.h:265:9: error: 'PortType' does not name a type
  265 |         PortType clkmask = 0;
      |         ^~~~~~~~
C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_STM32a.h:275:9: error: 'OE_PWM_Polarity' does not name a type
  275 |         OE_PWM_Polarity OE_polarity = OE_PWM_POSITIVE;
      |         ^~~~~~~~~~~~~~~

Used library Version Path
SPI          1.0.0   C:\Users\Phalguni\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.6.0\libraries\SPI
SrcWrapper   1.0.1   C:\Users\Phalguni\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.6.0\libraries\SrcWrapper

Used platform            Version Path
STMicroelectronics:stm32 2.6.0   C:\Users\Phalguni\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.6.0
Error during build: exit status 1

fpistm, 2023-08-17:
You build  for  a F4 not an F103 and errors comes from the library you used. Open an issue on this library.

closed 2023-08-17 by fpistm

fpistm, 2023-08-28:
For ref: https://github.com/board707/DMD_STM32/issues/46